robotology/mesh-support #1034: Connection errors to the torso

S-Dafarra, 2020-11-20:
## Brief description of the request/failure
After restarting the ``yarprobotinterface`` it often happens that the robot is not able to communicate with the torso. I attach here the log.

## Detailed context
There seems to be problems in communicating with the torso.

## Accompanying material

[log_icub-head_yarprobotinterface_26057_torso_not_Starting.txt](https://github.com/robotology/icub-tech-support/files/5575415/log_icub-head_yarprobotinterface_26057_torso_not_Starting.txt)

cc @GiulioRomualdi

S-Dafarra, 2020-12-09:
The robot is not starting anymore 
[log_icub-head_yarprobotinterface_2960_notStarting.txt](https://github.com/robotology/icub-tech-support/files/5664842/log_icub-head_yarprobotinterface_2960_notStarting.txt)

S-Dafarra, 2020-12-09:
By looking at the ``FirmwareUpdater`` it seems that the problem may be on the right arm. Indeed, we had to mount a new cover recently cc @salvi-mattia 

[image]

julijenv, 2020-12-09:
the cable from EB29 (right forearm - C shape MC4+) and the torso had the green cable broken from the root of the connection inside the upper arm - probably due to continuous open and close cover to check hardware pblm

closed 2020-12-09 by julijenv