robotology/mesh-support #1035: Right hand does not calibrate correctly

S-Dafarra, 2020-11-20:
## Brief description of the request/failure
The right hand does not complete the calibration successfully.

## Detailed context
[image]

## Accompanying material

cc @GiulioRomualdi

violadelbono, 2020-12-01:
Hi @S-Dafarra @GiulioRomualdi 

The problem of the hand shaking was solved by changing the PID parameters of wrist pitch and yaw of both of the arms. | Kp | was set from:

[image]

---

to the values currently set in the green iCubGenova04:

---

[image]

**HOWEVER** this change was made in the robot folder ($ROBOT_CODE/robotology-superbuild/robotology/robots-configuration/iCubGenova09) at branch `feature/wbd` so the folder needs to be pushed in the github repo and then re-compiled in order to send these changes to `/ICUBContrib`

Feel free to close this issue after the repo is updated.

S-Dafarra, 2020-12-01:
Hi @violadelbono, thanks a lot for that. The ``feature/wbd`` has been recently merged into ``devel``: https://github.com/robotology/robots-configuration/pull/213, so probably if you apply those modifications on a branch on top of ``devel``, you can already open a PR for that. In any case, I can also do that tomorrow.

S-Dafarra, 2020-12-02:
Ciao @violadelbono!, I updated the configuration files as you suggested and the left wrist does not oscillate anymore. On the other hand (literally), the thumb still does not calibrate. I guess the tendon is still broken, right?

violadelbono, 2020-12-02:
Yes, I will have time to check the finger probably tomorrow

S-Dafarra, 2020-12-07:
Ciao @violadelbono, @julijenv, do you have any ETA for this?

S-Dafarra, 2020-12-15:
Thanks @julijenv @violadelbono