robotology/mesh-support #1037: iCubGenova03 - PC104 not starting Power issue

gonzalezJohnas, 2020-11-23:
## Brief description of the request/failure
The PC104 is not starting possibly due to power issue.

## Detailed context
When the PC104 is stopped it fails to start, debugging the DEBIAN partition we can see that the OS don't start neither the BIOS.

## Accompanying material

julijenv, 2020-11-26:
FRom what I heard, it seems to be due to a dead battery BIOS, and it needs to be changed I have no idea how to do it, because I never face this kind of problem you are unfortunately the only one (maybe 2 but I'm not sure about that either) using the motherboard ATOM.
I recommend some video getting from internet to make the battery BIOS removable, but first you need to remove the one still installed on the head.
let us know

reafrancesco, 2021-01-21:
Hi!
I have news. I have changed the pc104 (i have bought another one with our budget) and the pendrive (brand new with the new image) and things work ONLY IF I powerup the pc104 when this is unplugged from the CFW2. These tests led me to conclude that the problem is the capacitors in the CFW2 that can degrade with the years (I guess @maggia80) can confirm this can happen (I have also talked to Claudio Lorini and he agrees this can happen). Please could we change the CFW2 otherwise we cannot use the  iCub Red (as it has been going on for months now). Do you have one that we can borrow?

maggia80, 2021-01-21:
hi @reafrancesco we should have spare CFW-002 available looking at the warehouse stock. We just need to check if we have it for real, but we can surely provide you a new one.

reafrancesco, 2021-01-21:
fantastic! thank you @maggia80

julijenv, 2021-01-25:
Hi @reafrancesco ,
We checked out the warehouse and we can confirm you that we have some CFW2 spare.
At the moment I dont know when we will be able to come to install it. Mostly on Friday, as we have the car. but maybe before depending on the schedule. The other possibility will be to have come here and get the board and install it.
let me know

julijenv, 2021-01-29:
Hi @reafrancesco ,
we finally nailed it.
The problem was due to a defective PSCPU.

IITCODE Qta Description
---|---|---
1243 | 1  |  PS_CPU-001, IIT - Power supply electronic board for CPU, Vin 12/14Vcc, Vout12V@500mA, 5.5V@10A

closed 2021-01-29 by julijenv