robotology/mesh-support #1039: iCubGenova04 right thumb oppose and proximal does not work

kouroshD, 2020-11-25:
## Brief description of the request/failure

The  right thumb oppose and proximal of iCubGenova04 does not work.
## Detailed context
One of the two coupled motors of right thumb works, but the other one does not work. So we can control well the robot right thumb finger.

## Accompanying material

kouroshD, 2020-11-26:
This morning, when I started the robot it was working the right hand. However, after some time all the fingers starts shaking and went to hardware fault again.

julijenv, 2020-12-04:
It has been restored the nentire wiring on the last finger.
it has been done the recalibration of all the joints of the forrearm
THe robot start nicely, I suppose everthing is alright now, let me know

kouroshD, 2020-12-07:
Thanks a lot @julijenv  for the great work you have done. 
I'll confirm that right thumb oppose and proximal are working correctly now.

closed 2020-12-07 by kouroshD