robotology/mesh-support #1040: Calibrate comprehensively the left and hand right of the ICubGenova04 hands

kouroshD, 2020-11-26:
## Brief description of the request/failure
Recently, we are intensively working on teleportation of the iCub hands, how in order the enable higher manipulation capabilities with iCub when teleoperating the robot. However, we are facing a lot of problems with the hardware and this causes problem. These problem appears both on left and rght hands of the robots.

## Detailed context
- On the right hand of the robot probably there is some short circuit. This causes the robot hand shakes time to time.
- `r_hand_fingers` and `l_hand_fingers` does not work most of the times and they do not cover their complete range of motions.
- The cabling and tendons of the robot needed to be calibrated and checked, so that we can cover the available mechanical range of motion since this can help for higher manipulation capability.
- However `r_pinky` and `l_pinky` mechanical structure allows for higher range of motion to close better, but this is not the case currently. By reaching the maximum value in the `yarpmotorgui`, we can only reach partially closed values.
- The hall sensors needed to be checked and calibrated, e.g. referring to https://github.com/robotology/icub-tech-support/issues/980
- The finger mechanical connection, especially the thumb connections are flimsy.

## Accompanying material

Hand Shaking:

[image]

Flimsy connections:

[image]

julijenv, 2020-12-04:
It has recalibrated both hand yesterday.
let us knwo what you think

DanielePucci, 2020-12-04:
CC @kouroshD

kouroshD, 2020-12-07:
Thank you a lot @julijenv  for this work. That was great. 
When I started the robot I had some issues:
- `r_pinky` did not start correctly. This has been ,mentioned in icub04 Teams chat as well. https://teams.microsoft.com/l/message/19:f8db55757ab7461c93622e40e5b5f273@thread.tacv2/1607028431687?tenantId=bf2f465e-1f75-4b6f-be29-bf53273bc1da&groupId=0ad0a501-ae56-494f-8425-9f5d08c9763f&parentMessageId=1606991166325&teamName=iCub-CRIS&channelName=iCubGenova04%20-%20Green%20iCub&createdTime=1607028431687

- `r_middle_ distal` does not work as well, stays at `0` and does not move.

As we checked on Friday morning with @julijenv, after adding changes to robot configuration files, they have not been installed before running `yarprobotinterface` on `iCub` when committed to `dic-iit/devel_iCubGenova04`. So, probably because of the that the two aforementioned joints did not work.

Besides those two issues, I have to say that thanks to the work done by @julijenv, I did not see anymore the misbehavior because of the short circuit.

Regarding the other points mentioned in the first comment, I did not manage to check them because of being short in time and needed to run the experiments.  Since, we will not have the robot for around 10 days, I would close this issue and if there is hardware issues I will consult with @julijenv , and if necessary reopen this issue.

Regarding the two broken joints, @julijenv please let me know if you prefer I open new issues specifically to them, or we use this issue for referring to them. Let me know please.

kouroshD, 2020-12-09:
> r_middle_ distal does not work as well, stays at 0 and does not move.

I tried this today, and it was working! So, the only problem is related to `r_pinky`.

julijenv, 2020-12-16:
It has been recalibrated eveyr fingers of r both left and right hand

closed 2020-12-16 by julijenv

kouroshD, 2020-12-17:
Thank you @julijenv

julijenv, 2020-12-30:
https://github.com/robotology/icub-tech-support/issues/997
I close also this one because it was related