robotology/mesh-support #1042: The wheels of the iCubGenova09 pole are very bent

S-Dafarra, 2020-12-03:
## Brief description of the request/failure
Several of the wheels of the iCubGenova09 pole seem to be about to fall (see picture below). As a consequence, it also became hard to move since the chassis closes the brakes.

## Detailed context
By looking at the back of the robot, I have noticed the two wheels on the right to be particularly bent.

## Accompanying material

[image]
[image]
[image]

cc @pattacini @GiulioRomualdi @prashanthr05 @DanielePucci

julijenv, 2020-12-03:
Hi @S-Dafarra ,
unfirtunately I cant do anything as long as the robot is on the stand, Plus my first question, will be: wwas the stand created to bear the weigh of iCub3? maybe iCub3 is way heavier than each wheel can stand.
Anyhow, net time you have the robot away form the stand letme know and I will check

DanielePucci, 2020-12-03:
@julijenv 
> unfortunately, I can't do anything as long as the robot is on the stand,

I believe you can use a crane to lift the robot

S-Dafarra, 2020-12-03:
The robot is held on the pole by three screws (hexagonal, dimension 5), accessible from the bottom of the root link. In case, I can help you tomorrow in taking it off the pole

S-Dafarra, 2020-12-04:
The screws holding the wheels were almost completely unscrewed. I tighten them up and the problem is fixed. Closing

closed 2020-12-04 by S-Dafarra