robotology/mesh-support #1048: Left hand thumb and index goes to not-configured

kouroshD, 2020-12-14:
When, we try to retarget the left hand of the iCub3,  some times it goes to not configured status. This is happening specifically for thumb and index proximal. This issue happened two times, both when the robot try to hold or grasp some object.  Moreover, when this happened, the `yarpMotorGui` was still showing those axes are working (staying with green color), however, they are not controlled. When we tried to put them in idle, they goes directly to `not Configured` status, and we should restart the `yarprobotinterface`.

## Brief description of the request/failure

## Detailed context

## Accompanying material

S-Dafarra, 2020-12-14:
> Logger did not provide any informative log, so, we are not reporting it here.

Namely, it saturated a long before.

pattacini, 2020-12-14:
Most likely the embedded boards get somehow stuck and/or reset. This can explain the behavior you've observed.

cc @julijenv

julijenv, 2020-12-16:
It has been recalbibrated both left and right hand. Hope this help.I never uncountered any failure during my 10 to 15 times relaunching the robto to recalibrate it.

julijenv, 2020-12-30:
After having tried with @kouroshD , the overload of the closing fingers put the board (MC4-PLUS) in an undrivable state.
I ask the DIC team to reduce the PID and PWM (or at least to investigate by attempts to find an optimal setup value) to prevent this phenomenon because as it happens on both arms, basically it means that the overload of the board hasn't been taken in account from the get go.

S-Dafarra, 2020-12-30:
The weird thing is that we had to reduce the maximum PWM by a good amount, making the thumb moving much slower. I wonder whether both boards could be faulty

kouroshD, 2021-01-04:
Due to the slow motion of the associated fingers at the moment,  we were discussing with @julijenv to reset the max pwm values to the previous values one by one in order to understand better where the problem is coming from. Is this action point is still relevant @julijenv ?

julijenv, 2021-01-04:
Yes, I believe it is still useful to investigate. I dont believe that much of a coincidence having both board (left and right faulty).
Let me know.

closed 2021-01-18 by julijenv