robotology/mesh-support #1050: icub3 left arm little fingers do not calibrate

kouroshD, 2020-12-18:
## Brief description of the request/failure
icub3 left arm little fingers do not calibrate. The problem seems to be  from mc4 C-shape eb29 board, that the is not reading the feedback values. @julijenv  investigated the issue and tried to fix it, but apparently the problem is from the board and we may need to substitute it.

## Detailed context
```
 <ERROR> Right_Arm_Calibrator cannot perform 'check hardware fault' operation because joint number 15 is out of range [].
```

## Accompanying material

[log_icub-head_yarprobotinterface_1437.zip](https://github.com/robotology/icub-tech-support/files/5717200/log_icub-head_yarprobotinterface_1437.zip)

julijenv, 2020-12-18:
Hi @kouroshD ,
after chekcing for some time with @S-Dafarra and @kouroshD , it looks like the C shape MC4+ board has a problem or at least it needs to be tried.

cc @violadelbono , could you look at thtat ?

S-Dafarra, 2020-12-18:
Note, it is the **Right** arm to have this problem.

DanielePucci, 2021-01-04:
Hi @julijenv @violadelbono, let us know when you may process this issue

julijenv, 2021-01-04:
I will be back on Monday (Im currently in quarantine).
If @violadelbono dont have any other things plan when she will return on the 7th if I recall well, maybe she could try to change the board C shape and C, or else it will be done next Monday WCS

violadelbono, 2021-01-07:
Working on it! @DanielePucci

violadelbono, 2021-01-08:
Hi @kouroshD and others,

Good news! 🎉  The problem is solved and it was actually easier than it seemed.
It was not related to the board itself, but to an error in the syntax of the calib file of the right arm (`left_arm-calib.xml` inside calibrators folder). In particular I noticed something strange in the calibration order parameter (extra space/parenthesis related to the last joint -a typo I suppose), so I corrected it and now it looks like the default: 

      <param name="CALIB_ORDER">(0 1 2 3) (4) (5 6) (7) (8 9 11 13) (10 12 14 15) </param>

For the other arm (left) I momentarily excluded from calibration order joints  8 9 10 because of issue [1051](https://github.com/robotology/icub-tech-support/issues/1051)

Side note: the error you reported in the issue description is the exact error occurring when a joint is excluded from the calibration sequence (good to know for everyone for the future 😃 )

S-Dafarra, 2021-01-08:
> Hi @kouroshD and others,
>
> Good news!  The problem is solved and it was actually easier than it seemed.
> It was not related to the board itself, but to an error in the syntax of the calib file of the [...]

Great @violadelbono! I guess the error message was a bit misleading then :sweat_smile:

closed 2021-01-11 by violadelbono