robotology/mesh-support #1051: Left thumb oppose broken

S-Dafarra, 2020-12-18:
## Brief description of the request/failure
After a hit against one of the robot legs, the right thumb oppose cable got broken

## Detailed context

## Accompanying material

S-Dafarra, 2020-12-18:
cc @kouroshD @DanielePucci

julijenv, 2020-12-30:
I do believe it is the one I restore before leaving.

S-Dafarra, 2021-01-04:
@julijenv I believe you fixed the right one https://github.com/robotology/icub-tech-support/issues/1035#issuecomment-745380367

This one broke after you left.

julijenv, 2021-01-04:
> @julijenv I believe you fixed the right one [#1035 (comment)](https://github.com/robotology/icub-tech-support/issues/1035#issuecomment-745380367)
> 
> This one broke after you left.
:sob:

julijenv, 2021-01-11:
Hi @kouroshD , @DanielePucci , @S-Dafarra ,
TO resolve this issue, we need at least **2 days** - because unfortunately one part of the thumb is damaged and we will need to recable, rewired and replace this piece. (3 tendons to recable + piece to replace + electronical wires to rewires)+ time to remove everything).
Let us know when it is convenient for you to start doing it and when it is not cumbersome in your experiments.
let us know
cc @violadelbono

S-Dafarra, 2021-01-11:
> Hi @kouroshD , @DanielePucci , @S-Dafarra ,
> TO resolve this issue, we need at least **2 days** - because unfortunately one part of the thumb is damaged and we will need to recable, rewired and [...]

Hi @julijenv, thanks for letting us know. For sure, we would need the robot on Wednesday and Thursday. Tomorrow the robot will be already on maintenance to install the skin. Do you think it may be doable between today and tomorrow?

julijenv, 2021-01-11:
Hi @S-Dafarra ,
as explained in my previous mail we need 2 days so it is more realistic to think that if we take it now we will be able to give it back to you on wednesday afternoon.
if it is fine by you, we will start ASAP

S-Dafarra, 2021-01-11:
@pattacini @Fabrizio69 do you think that this activity may interfere with the skin installation and configuration?

pattacini, 2021-01-11:
Maybe if @julijenv will remove the lower arm entirely to repair the thumb there won't be any "conflict" with mounting the skin on the upper arm. Though, I don't know if for testing purposes @simeonedussoni will need to have the whole arm chain in place.

pattacini, 2021-01-11:
Just talked with @simeonedussoni and I believe we can proceed as I suggested above.

For a quick test, indeed, @simeonedussoni could connect the CAN box to the MTB (no need to have the whole arm in place).
Once the thumb will be repaired and thus the lowe arm back in place we will be finalizing the tests with the `yarprobotinterface`.

S-Dafarra, 2021-01-11:
Thanks @pattacini. Then @julijenv, green flag for proceeding for me. I prefer to start the tests with the robot fully functional.

julijenv, 2021-01-14:
It has been restored every single cable from the thumb as well rewire the hall sensor of the base of the thumb as we had to remove the first phalange to repair a knot screw broken.
it has been taken from the warehouse the following items

IITCODE| Qta |DEscription
---|---|---
1815|2m|carlstahl 0.63mm non coated
2391|1m|carlstahl 0.61mm coated
706|1|hall sensor

closed 2021-01-14 by julijenv