robotology/mesh-support #1052: iCubOS image request

Iaxama, 2021-01-11:
## Brief description of the request/failure
Given the necessity to reinstall the OS on the laptops of the iCub Purple robot, we would need the iCubOS images for either laptops, CUDA workstation, head and server.

Iaxama, 2021-01-11:
@traversaro

traversaro, 2021-01-11:
cc @mbrunettini 

Note that this request has been done following the documentation in https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-from-image/#where-to-get-the-latest-icubos-image .

julijenv, 2021-01-14:
Hi @mbrunettini 
in proto we have one head already cabled (ex-TUC) that can be taken to try to mount the new OS if i can be of any help..
@Fabrizio69 , @violadelbono

Iaxama, 2021-01-14:
In the end we have managed to set up the OS starting from a clean Ubuntu 20.04 installation. The server and the head are unchanged, and the network set up was pretty smooth. First tests show that the regular workflow to start up the robot is working as usual.

traversaro, 2021-01-18:
> 
> 
> Hi @mbrunettini
> in proto we have one head already cabled (ex-TUC) that can be taken to try to mount the new OS if i can be of any help..
> @Fabrizio69 , @violadelbono

To clarify, the request from @Iaxama was about the latest image with Ubuntu 18.04, not the new one (that I guess is still not ready) with Ubuntu 20.04 .

violadelbono, 2021-03-12:
@Iaxama can we close this issue?

closed 2021-03-12 by Iaxama

traversaro, 2021-03-12:
For everyone from the community still interested on this: the uploading of the iCubOS image somewhere from which anyone can download them is still a work in progress, but it is tracked in an IIT internal repo.