robotology/mesh-support #1053: iCubGenova01 - Left hand thumb proximal not providing enough power

xEnVrE, 2021-01-11:
## Brief description of the request/failure

I am trying to close a control loop with tactile sensors using PWM as control output. While I am able to obtain satisfactory performance with the index and middle proximal joints, the thumb proximal (while being able to move with no load) seems not to respond with enough power while under load.

## Detailed context

@julijenv already checked the thumb proximal wire and noticed that there is something wrong.

julijenv, 2021-01-12:
Hi @xEnVrE ,
one knot srew was leaving the cable slipped away , I think i succeed to block it, please make some trials and let me know.

xEnVrE, 2021-01-22:
Hi @julijenv,

after some experiments, involving grasping, the same problem occurred.

As further description, when I try to operate the thumb opposition from the `yarpmotorgui` in `POSITION` mode, it cannot go beyond 50 degrees and I can hear the motor spinning with no actual motion of the joint.

violadelbono, 2021-01-25:
Hi @xEnVrE 

We blocked the cable again as a temporary quick solution. Hope it will last for your next experiments 🤞

xEnVrE, 2021-01-25:
Thank you @violadelbono @julijenv

julijenv, 2021-02-04:
Hi @xEnVrE , It has been retensed periodically, the main problem is that the screw nut holding firmly the proximal cable is not well placed upon the main throat, leaving the cable to slide.

julijenv, 2021-02-18:
Can we consider this issue closed?
thx i advacnce
@xEnVrE

xEnVrE, 2021-02-18:
I think we consider it closed. In case I'll re-open it :)

closed 2021-02-18 by xEnVrE