robotology/mesh-support #1054: The robot power supply makes a lot of noise when the boards are running

S-Dafarra, 2021-01-13:
## Brief description of the request/failure
After the new power cable has been mounted on iCubGenova09, we noticed that the power supply makes an annoying whistle when starting the motors. It was not happening before.

## Detailed context

## Accompanying material

julijenv, 2021-01-15:
Hi @S-Dafarra ,
unfortunately the sound is coming from the reaction from the Power supply to the plug of the current sensing that has been added. It regulates the voltage and current and the frequencies of the pwm can be inside the hearing bandwidth, the only thing that can be done is removing it to get rid off the noise.
let me know what you prefer.

S-Dafarra, 2021-01-15:
> Hi @S-Dafarra ,
> unfortunately the sound is coming from the reaction from the Power supply to the plug of the current sensing that has been added. It regulates the voltage and current and the [...]

What do we gain in having the current sensing wrt not having it?

S-Dafarra, 2021-05-24:
We got used to it :grin: Closing

closed 2021-05-24 by S-Dafarra