robotology/mesh-support #1057: iCubGenova01 - pc104 is not reachable

xEnVrE, 2021-01-14:
## Brief description of the request/failure

While using the robot, it suddenly stopped streaming data from the `pc104` outwards.

After that, we tried to reset the pc104 but we cannot reach it anymore (e.g. the destination is unreachable while trying to ping it)

cc @vtikha

xEnVrE, 2021-01-14:
After some trials, we were able to get inside the pc104 and run the `yarprobotinterface`: the robot started as usual.

However, the `yarpdev` responsible for the cameras do not start.

```
icub@pc104:~$ yarpdev --from camera/ServerGrabberDualDragon.ini --split true
[WARNING] |yarp.device.grabberDual| Period parameter not found, using default of 0.03 s
[INFO] |yarp.os.Port| Port /icub/cam/left/rpc active at tcp://10.0.0.2:10077/
[INFO] |yarp.os.Port| Port /icub/cam/left active at tcp://10.0.0.2:10078/
[INFO] |yarp.os.Port| Port /icub/cam/right/rpc active at tcp://10.0.0.2:10079/
[INFO] |yarp.os.Port| Port /icub/cam/right active at tcp://10.0.0.2:10080/
[DEBUG] Format = 1
[ERROR] no active cameras
[ERROR] LINE: 336
DragonflyDeviceDriver2: can't open camera
[ERROR] |yarp.dev.PolyDriver| Driver <dragonfly2> was found but could not open
[DEBUG] Format = 1
[ERROR] no active cameras
[ERROR] LINE: 336
DragonflyDeviceDriver2: can't open camera
[ERROR] |yarp.dev.PolyDriver| Driver <dragonfly2> was found but could not open
[ERROR] |yarp.device.grabberDual| Opening devices... FAILED
[ERROR] |yarp.device.grabberDual| Error while opening subdevice
[ERROR] |yarp.dev.PolyDriver| Driver <grabberDual> was found but could not open
[DEBUG] |yarp.dev.Drivers| ===============================================================
[DEBUG] |yarp.dev.Drivers| == Options checked by device:
[DEBUG] |yarp.dev.Drivers| ==
[DEBUG] |yarp.dev.Drivers| device=grabberDual
[DEBUG] |yarp.dev.Drivers| ==
[DEBUG] |yarp.dev.Drivers| ===============================================================
[ERROR] |yarp.dev.Drivers| yarpdev: ***ERROR*** device not available.
[INFO] |yarp.dev.Drivers| Suggestions:
[INFO] |yarp.dev.Drivers| + Do "yarpdev --list" to see list of supported devices.

```

Does `[ERROR] no active cameras` mean that the firewire boards are not detected?

julijenv, 2021-01-20:
Hi @xEnVrE ,
we made a crupolous check of the wire inside the head of iCubGEnova01 nad could not find anything wrong, we remove the dust that was stuck inide the fans and fac expression board.
we checked that the telecamera were working and try to challenge the flat cable to detect a worn one, but couldnt.
it has been changed 2 BLL boards.

closed 2021-01-20 by julijenv

xEnVrE, 2021-01-20:
Thank you @julijenv