robotology/mesh-support #1058: left and right hand thumb opposition and proximal does not work well iCubGenova09

kouroshD, 2021-01-15:
## Brief description of the request/failure

When I start the yarprobotinterface of  iCubGenova09, the left thumb does not calibrate well, so that the left thumb and index finger get in contact. After the calibration is finished, we cannot move the left and right thumb proximal and opposition. 

## Detailed context
I checked the analog and ecnoder values, they seems to me OK.

## Accompanying material

I checked the logger, only complaining about the harware limit reached, when it goes to an ill pose.

julijenv, 2021-01-18:
Hi @Kourosh,
did you make some modification on the PID of the thumb oppose by an chance?

julijenv, 2021-01-18:
#1048, idem

kouroshD, 2021-01-18:
I did that back in time to fix the problem of that related issue and it was working although with slower motion. I did not change them after that point, but I can revert them back to the original values to see if they work.

kouroshD, 2021-01-18:
I reverted back to the original values #1048. All the motors configures, however, after the calibration is done, I again could not move the thumbs in the expected way when I give command through yarpmotorgui. Some times they move, sometimes not.
Moreover, in the previous calibration process, l_index_distal was not configured, but the try it was configuring. I did not changed the pwm values for this axis.

kouroshD, 2021-01-18:
Here is the version of the firmwares.
[image]

cc @S-Dafarra

kouroshD, 2021-01-18:
I did a new trial with the yarprobotinterface again:

[image]

[image]

violadelbono, 2021-02-04:
Hi @kouroshD, do you think we can close this issue?

S-Dafarra, 2021-02-04:
I think that the right thumb opposition reductor is still broken, right @julijenv?

kouroshD, 2021-02-17:
The right thumb cable is broken at the moment.
@julijenv  has fixed the problem with the `l_middle_proximal` while checking the hands.

violadelbono, 2021-03-29:
@GiacomoVassallo13 if you please can check both hands and repair the broken cables

S-Dafarra, 2021-04-16:
I think that the right thumb oppose still has issues. It often gets stuck in the zero position, and it is needed to move it by hand. Also, it often fails to calibrate.

kouroshD, 2021-04-30:
@violadelbono  @GiacomoVassallo13  Is there is any update on fixing this issue? I was trying to use the robot, and I can confirm that there is a problem with the right thumb.

CC @DanielePucci  @S-Dafarra

violadelbono, 2021-05-03:
@kouroshD I booked the robot for this afternoon

violadelbono, 2021-05-03:
Good news: the motor, sensor and wiring are fine. In fact, when trying to move the joint with the jogger, it works perfectly.
Therefore, I believe it is a calibration problem in terms of the magnet position. I tried to move the magnet but didn't succeed.  I will ask for support tomorrow since it is in a very tricky position :( 

I will update you tomorrow morning.

@kouroshD @S-Dafarra

S-Dafarra, 2021-05-03:
> Good news: the motor, sensor and wiring are fine. In fact, when trying to move the joint with the jogger, it works perfectly.
> Therefore, I believe it is a calibration problem in terms of the magnet [...]

Nice @violadelbono, thanks for that. We were able to move it also from the motorgui, but only when the thumb proximal and distal are far from the zero position. Otherwise, they are stuck. I wonder if it is the same as https://github.com/robotology/icub-tech-support/issues/1089

violadelbono, 2021-05-04:
Hi @S-Dafarra @kouroshD,

We believe the problem could be related to the joint 9 (thumb proximal) since its pulley, and thus the joint, is shaking when the motor of j8 is moving.

Unfortunately the entire forearm needs to be removed in order to check the proximal thumb motor + pulley.

Give us the green light and I will book the robot when you are not using it.

cc @gandoo @elio75

https://user-images.githubusercontent.com/61685519/116991310-a539b980-acd4-11eb-8587-ca34793ce1b7.mp4

https://user-images.githubusercontent.com/61685519/116991388-c4384b80-acd4-11eb-8da0-fc57fe8a4954.mp4

S-Dafarra, 2021-05-04:
Yes, we can try to schedule this soonish. What would be your proposal? Let me put in the loop also @pattacini and @davidetome.

pattacini, 2021-05-04:
We're mostly working on the lower body, so go ahead and take the robot as you wish.
We will adpat.

violadelbono, 2021-05-04:
@S-Dafarra I propose Thursday morning

S-Dafarra, 2021-05-04:
> @S-Dafarra I propose Thursday morning

Sounds good then!

violadelbono, 2021-05-06:
I rewired the thumb proximal joint with a new cable and I noticed that:

- the cable was overlapped on the pulley
- the cable was entering the pulley frame in the wrong slot:

[image]

This could have caused friction and a bad angulation of the cable, leading to the thumb wobbling. However I am not sure this solved completely the problem. 
Unfortunately I was not able to try the robot yet with the motorgui because they are still working on the legs.
I will try the thumb as soon as the robot is fully assembled.

S-Dafarra, 2021-05-06:
> Unfortunately I was not able to try the robot yet with the motorgui because they are still working on the legs.
> I will try the thumb as soon as the robot is fully assembled.

If it is just a matter of starting the robot without some parts, I can help you with that.

violadelbono, 2021-05-06:
@S-Dafarra I tried but got a lot of errors. Probably because the leg boards are still attached and powered but all collected in a plastic bag 🤣 so they could create some short circuit idk

S-Dafarra, 2021-05-06:
Ah too bad. You were receiving errors also from the torso and the arms then?

violadelbono, 2021-05-06:
Yes, including fault button pressed even though it was not 🤔 I checked the right arm cables and they look fine. Bo, I will investigate more

Gandoo, 2021-05-07:
ciao @S-Dafarra as it says @violadelbono  do not turn on the robot because all the boards of the legs are removed from the supports and if they come into contact they make a short circuit. At that point we can put the robot in the bin of the undifferentiated !!

S-Dafarra, 2021-05-07:
We could try to recycle some of it 😁 In any case, roger that, let me share it also on the Teams channel then!

violadelbono, 2021-05-18:
Hi @kouroshD and @S-Dafarra,

You can try to use the thumb and tell me if it got better after the re-wiring. Thanks

kouroshD, 2021-05-21:
Here are the results of an experiment I performed. Moreover, I have compared the results with the left hand to see the performance difference.

In general, as I see still the problem exist and tracking has poor performance.

|motor| right hand (low performance)| left hand|
|---|---|---|
| thumb oppose | [image]   |  [image]  |
| thumb proximal | [image]  |  [image]  |
| thumb distal |  [image]   | [image] |

The problems are with the right hand:
- the thumb oppose and proximal of the right hand still does not track _well_ the reference.
- The  thumb distal is not tracking well the reference. In fact, by touching the **thumb middle joint**, it is easily observable that there are high friction when moving, or the motion of that **joint** is not smooth, and probably this is the reason behind the poor tracking of motor thumb distal motor.

Moreover, the right middle distal motor several times went to HW fault.

Uboldi80, 2021-09-28:
i think this problem has fixed in: https://github.com/robotology/icub-tech-support/issues/1198

and after the tensioning of the thumb tendon.

pattacini, 2021-09-28:
cc @kouroshD

kouroshD, 2021-09-28:
Thank you @Uboldi80 for the fix.

Uboldi80, 2021-09-28:
Nothing @kouroshD  but before to close the issue i suggest to try again the joint to verify if is acceptable.

THKS.

kouroshD, 2021-09-28:
Yes, for sure. I will try it in the next coming days and will let you know. Thanks.

pattacini, 2021-09-28:
Just a quick comment: this issue was also related to the status of the left-hand fingers, whereas the last intervention was on the right thumb.

At any rate, quite some time has passed since the opening, thus it might be a good move to give the hands a try from the application standpoint anyway, @kouroshD

cc @Uboldi80

Uboldi80, 2021-09-29:
for some reason that I still do not fully understand, despite the pulling of the hand tendons being performed at least 2 more times, the tendon retaining grains were again slack.
this obviously contributes to making the joint difficult to control as well as making it oscillate.
This morning I re-tensioned the thumb cables but this time I added a special low-resistance bellow to prevent damage by irreparably blocking the grain.
it is loctite 222 pink.
probably part of the problem is due to the size of the fixing screws: I believe that M 1.6 is really too little to offer adequate resistance, also considering that the threaded piece I believe is in aluminum, so I assume that following some uses of this joint in particular (thumb) subjected to greater stress than the other joints tends to release the retaining grain in a short time.
it would probably help if the grains of the fingers were at least M2 in order to guarantee a better seal against the tendon.

Uboldi80, 2021-11-02:
@isorrentino 
please, let me know if the problem is fixed or improved, after the maintenance intervention post maker faire.

thks!!

S-Dafarra, 2021-11-10:
Hi @Uboldi80. I think so, also because the history of this issue is getting long. I think we can close it and if we need we can open new ones.

closed 2021-11-10 by pattacini