robotology/mesh-support #1060: iCubGenova02 hardware check

Iaxama, 2021-01-18:
## Brief description of the request/failure
iCubGenova02 hardware check

julijenv, 2021-01-18:
Hi @Iaxama ,
It has bben decded together with you and @2103simon to start with the following list of check:
- make the robot start correctly from the get go (2nd attempt after the head has been started).
- tense everything that looks loose
- make fingertip sensor working
- give the datasheet of fans (backpack and head)

Iaxama, 2021-01-18:
@ChiaraBartolozzi

julijenv, 2021-01-20:
fans in the head
[image]
i 3 sono uguale e sono questo codice
IITCODE| ALias |DEscription
---|---|---
5797 | ebm-papst_255m | Fan 5VDC, screw mount, 25x25x8mm

datasheet
[5797.pdf](https://github.com/robotology/icub-tech-support/files/5843620/5797.pdf)

violadelbono, 2021-01-20:
> - give the datasheet of fans (backpack and head)

Backpack:
- 1 & 2:  IITCODE 5797 (same one of the head, see above for datasheet)

- 3: 

| IITCODE  |  ALIAS | DESCRIPTION  |  DATASHEET |
|---|---|---|---|
| 6259  | nmb-tech_1204kl-04w-b30-b00  | Fans	Fan 12VDC, screw mount, 30x30x10mm	  | [6259.pdf](https://github.com/robotology/icub-tech-support/files/5843692/6259.pdf) |

[image]

2103simon, 2021-01-20:
Thank you for the update!

julijenv, 2021-01-20:
- [ ] make the robot start correctly from the get go (2nd attempt after the head has been started).
at the second attempt to launch the robot:
here are the errors messages fro logger:
- paramneter networks use deprecated syntax - to be corrected
- strain timed-out board 10.0.1.3 right-arm
- 2FOC encoder phase broken 10.0.1.3 j0_3, part16 = 0x0000
- torso is not responsiveelbow cable loose - left arm
- retense, check or recalibrating the left_Arm: j9, j10, j14, 
- [ ]  tense everything that looks loose (mechanicals problems):
- distal medio left arm
TO BE ADDED - calibration j15 both arms

2103simon, 2021-01-20:
Any news about the plan regarding deep check?
@ChiaraBartolozzi

marco-monforte, 2021-01-26:
Hey everyone, reporting here some issues I had when starting the robot.

First time, right arm not started with the following log (I reported only the Errors):
[right_arm_error_log.txt](https://github.com/robotology/icub-tech-support/files/5873518/right_arm_error_log.txt)

After killing the yarprobotinterface the robot didn't move back to the resting position and, when I started it again, it didn't work and I got the following error log:
[startup_error_log.txt](https://github.com/robotology/icub-tech-support/files/5873564/startup_error_log.txt)

This last operation failed 2/2 times.

Hope it can be useful somehow (?)

violadelbono, 2021-02-02:
## 1st trial

I launched yarprobotinterface without motor reset for three times and the robot started at the third time:

-  NO START [first_start.txt](https://github.com/robotology/icub-tech-support/files/5912192/first_start.txt)
- NO START [second_log.txt](https://github.com/robotology/icub-tech-support/files/5912195/second_log.txt)
- START [third_successful.txt](https://github.com/robotology/icub-tech-support/files/5912201/third_successful.txt) some errors due to right shoulder out of limits 
- After reset of motors: zero errors from log file and all joints correctly calibrated

## 2nd trial

I turned off the head and tried again:

- NO START [first_start2.txt](https://github.com/robotology/icub-tech-support/files/5912304/first_start2.txt)
- START except right shoulder HW fault [second_log2.txt](https://github.com/robotology/icub-tech-support/files/5912333/second_log2.txt)
- START except right shoulder HW fault [third_log2.txt](https://github.com/robotology/icub-tech-support/files/5912371/third_log2.txt)
- After motor reset: same situation as above

## 3rd trial

After resetting the head:

- NO START [first-start3.txt](https://github.com/robotology/icub-tech-support/files/5912470/first-start3.txt)
- START NO ERRORS all joints calibrated

## 4th trial

After resetting the head:

- NO START [first_start4.txt](https://github.com/robotology/icub-tech-support/files/5912582/first_start4.txt)
- START NO ERRORS all joints calibrated

## 5th trial

Same situation as 4th trial.

2103simon, 2021-02-02:
@julijenv and @violadelbono, go for a deep check as soon as possible is my first idea when I read this... But it looks like it starts now after the first trial without errors, which is also nice and better than befor. Have you any feedback if a deep check is planned by the PIs as long iCub is not used for experiments? I have heard nothing.
Cc @ChiaraBartolozzi

Iaxama, 2021-02-03:
Thanks Viola, I have also struggled to see a pattern in the errors. Also I have noticed that the errors become more likely to happen if the robot is on for a long time (> 30 min) and then you try a restart. Maybe due to overheating??

violadelbono, 2021-02-04:
## Summary of errors

### 1st)

  - torso board
  
        <ERROR> theNVmanager::Impl::ask() had a timeout for BOARD torso-eb5-j0_2 IP 10.0.1.5 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 

        <ERROR> theNVmanager::Impl::wait() had a timeout for BOARD torso-eb5-j0_2 IP 10.0.1.5 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult

        <ERROR> Driver <embObjMotionControl> was found but could not open

- right arm board same error

       <ERROR> theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb4-j4_15 IP 10.0.1.4 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion 

- left leg same error

      <ERROR> theNVmanager::Impl::ask() had a timeout for BOARD left_leg-eb6-j0_3 IP 10.0.1.6 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion

      <ERROR> theNVmanager::Impl::ask() had a timeout for BOARD left_leg-eb7-j4_5 IP 10.0.1.7 and nv ID32 = 0x00000002 -> IND = 0, TAG = eoprot_tag_mn_comm_status_managementprotocolversion

 second time NO START: 

- torso board same error
- right arm same error
- left leg same
- IMU bosh

      <ERROR> Driver <imuBosch_BNO055> was found but could not open
- segmentation fault

      <ERROR> yarprobotinterface intercepted a segmentation fault caused by a faulty plugin:
      <ERROR> /usr/local/src/robot/icub-main/build/lib/iCub/cartesiancontrollerserver.so(_ZN25ServerCartesianController9detachAllEv+0x32) [0x7fc70eb2cf42] 

- encoder right arm

       <ERROR>  from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 573s 288m 706u: (code 0x0200000d, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + . 

- face

      <ERROR> FaceV3_Calibrator : Timeout while calibrating  0 
      <ERROR> FaceV3_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit 
      <ERROR> embObjMotionControl::checkRemoteControlModeStatus(0, 7103593) is unable to check the control mode of BOARD face-eb22-j0 IP 10.0.1.22 because it is now in HW_FAULT 

### 2nd)

- IMU bosh 

      <ERROR> First read from the device failed, check everything is fine and retry 
      <ERROR> A thread failed to start with error code: 0 
      <ERROR> Driver <imuBosch_BNO055> was found but could not open 

- encoder right arm + HW fault

### 3rd)

- IMUbosh
- segmentation fault

### 4th - 5th)

- IMU bosh

## To check

- [ ] Torso board eb5
- [ ] Right arm board eb4
- [ ] Left leg board eb6 - eb7
- [ ] Encoder right arm eb3 j0 
- [ ] Face eb22 j0 HW fault
- [ ] IMU bosh -known issue
- [ ] yarprobotinterface segmentation fault -

violadelbono, 2021-02-04:
>  - [x] Face eb22 j0 HW fault

There was friction in the eyelids. They have been polished and the PID values were changed. They've been also recalibrated.
All of this was done in .local/share/yarp/robots/iCubGenova02 so it will need to be committed on robots-configuration

julijenv, 2021-02-08:
Hi @Iaxama,
it has been checked and fixed the following points
 [x] Torso board eb5, right arm eb4, left leg eb6-eb7: check ETH connection
- It ahs been reflashed Loader, Updater and Apllcation on the board , it seems that the Laoder most probably was sometimes not getting throught the application and was not makgin the ETH bridage anymore.
 [x] Encoder right arm eb3 j0
When opened the chassis and remove the cover of the back pask it ahas been discovered that 2 crimp insde 2 connector were out or almost not touching, pull them in succeed to remove the error message of this arm.
 [x] Face eb22 j0 HW fault
It ahs been updated both PID and redefine the limit of the eyelids, now it works perefctly fine.

violadelbono, 2021-02-08:
Joint 15 left hand was broken. It has been repaired.

| IITCODE  |   QUANTITY| DESCRIPTION  |   |   |
|---|---|---|---|---|
| 2391  | 50cm   | Stainless steel microcable, cable construction 7x19mm, nom. dia 0.45mm, outer dia 0.61mm, Polyamide coated, AISI 316  |   |   |

julijenv, 2021-02-11:
I just finished reattaching the parts of the left arm cover
It is now possible to test the robot
let me know if the robot starts correctly ( it should now being able to start from the second attempt when the head starts - and as soon you start the motor ON, when the robot has already being used before.)

julijenv, 2021-02-15:
IT appears that the EMS board IP 10.0.1.4 seems to not boot correctly.
We are going to attempt to change it ad see if it solved the issue

Iaxama, 2021-02-15:
After many attempts we managed to run yarprobotinterface.
Even if I am not sure that was related to the error we were getting, we have updated the firmware of all the strain2 boards in the legs. 
Also we have got rid of anything related to the IMU in the icub_all.xml configuration file. Making it work probably requires installing drivers on the new version of the head, but that's another issue.

Even on a successful run of the robot motors we still get the following error from boards 10.0.1.2 and 10.0.1.4

`ERROR		embObjMC BOARD left_arm-eb2-j4_15 (IP 10.0.1.2)  Jointsset  0 hasn't joints!!! I should be never stay here!!!
`

julijenv, 2021-02-15:
to @marcoaccame ,
could you pleasse try to tell us what this error message represents? I never see it before!
let us know

lunagava, 2021-02-17:
yarprobotinterface is not still working, so I attached the log session file it has generated

[yarprunlog_17_02_2021_15_42_11.log](https://github.com/robotology/icub-tech-support/files/5996431/yarprunlog_17_02_2021_15_42_11.log)

marcoaccame, 2021-02-17:
> to @marcoaccame ,
could you pleasse try to tell us what this error message represents? I never see it before!
let us know

Hi @julijenv & @violadelbono, me neither: it is the first time I come across this error message.

...

After a quick research I found that the error is emitted by icub-main and in particular by function `bool embObjMotionControl::updatedJointsetsCfgWithControlInfo()`
when ... it detects 0 joints in a jointset:

```C++
    for(size_t s=0; s<_jsets.size(); s++)
    {
        if(_jsets[s].getNumberofJoints() == 0)
        {
            yError() << "embObjMC"<< getBoardInfo() << "Jointsset " << s << "hasn't joints!!! I should be never stay here!!!";
            return false;
        }

```

I don't know who wrote this check and for which reason. We need to investigate. For what I can see the check just prints an error and does not produce any harm to the system. We should manage this problem as a bug to be solved, with proper analysis and test.

Some hints.

I suspect that some xml pieces may be missing.... Yes: I don't see the section `JOINTSET_CFG` in file [`left_arm-eb2-j4_15-mec.xml`](https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/hardware/mechanicals/left_arm-eb2-j4_15-mec.xml) which for instance is in other similar files (see [here](https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/hardware/mechanicals/left_arm-eb1-j0_3-mec.xml#L67)).

So, maybe adding this section in the xml files could solve, but again: it is bug in `icub-main` that must be solved.

julijenv, 2021-02-17:
Hi @marcoaccame ,
actually as you saiid it, the robot started correctly but we had this error message that we couldnt figure out what is the purpose, thx for reassuring us.

julijenv, 2021-02-18:
if ever the proble witht he board 10.0.01.3 reappeared , be prepared to changed it.
in the meanwhile i close the issue

closed 2021-02-18 by julijenv