robotology/mesh-support #1062: iCub PC104 (atom) live Installation

reafrancesco, 2021-01-19:
I would like to kindly ask for the procedure for preparing the pendrive to start the PC104 atom.
Last time I discussed with @mbrunettini about our realtime live image I remember it was available from a link in the manual.
Now the link is not longer connected to resource with all the the icub live installations.
I have also tried to look into link I found in http://wiki.icub.org/iCub/downloads/pc104-images/ but also this link does not work.
@mbrunettini can you help us find the iCub live images (realtime) are?
To update also @julijenv with this, the reason why we would like to clone another pendrive is that we are not sure the problem on the PC104 power supply is linked to the pendrive.
Thank you so much for the support.

P.S.: we have been made aware that this request is similar to /robotology/icub-tech-support#1052 (#1052)

mbrunettini, 2021-01-20:
The correct link to download the image is :
[http://www.icub.org/download/pc104-images/](http://www.icub.org/download/pc104-images/)

You can follow the _new_ guide [here](https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/icub-live/)

violadelbono, 2021-03-29:
Hi @reafrancesco, can we close this issue? Everything ok?

closed 2021-04-14 by maggia80