robotology/mesh-support #1065: iCubGenova01 - issues with the right index finger

xEnVrE, 2021-01-22:
## Brief description of the request/failure

- the right index distal moves correctly but it cannot goes beyond ~30 degrees when commanded towards 0.0 degrees
- the right index proximal does not move at all and the encoder reading jumps between ~40.0 and 90.0 degrees

violadelbono, 2021-01-25:
> the right index distal moves correctly but it cannot goes beyond ~30 degrees when commanded towards 0.0 degrees

The joint was re-calibrated correctly

> the right index proximal does not move at all and the encoder reading jumps between ~40.0 and 90.0 degrees

This was due to a loose cable, which was tensed again. Moreover, the Hall sensor wires were making a contact with the robot frame, thus the encoder readings were jumping.

closed 2021-01-25 by violadelbono

xEnVrE, 2021-01-25:
Thank you @violadelbono