robotology/mesh-support #1066: iCubErzelli01: Left thumb broken

ddetommaso, 2021-01-26:
## Brief description of the request/failure
The left thumb oppose cable broke during the experiment.

## Detailed context

## Accompanying material

https://user-images.githubusercontent.com/3440640/105845207-b0e15a00-5fda-11eb-90b5-77f14430a38b.mp4

julijenv, 2021-01-26:
Hi @ddetommaso ,
just a quick look just let me see taht just the distal is broken, the thumb oppose looks ok but maybe a lttle loose or it is the proximal joint , hard to say they are all togehter bundle so will see.
I'm getting feedback to know if I could pass by tomorrow.

julijenv, 2021-01-29:
Hi @ddetommaso ,
It has been replaced the left thumb proximal and distal cable.
After having recabled and tested the robot we have noticed that also the rightthumb distal was broekn and it has been consequently replaced.

It has been taken from MM toolbox

IITCODE|Qta|Description
--|---|---
1895|1m|carlstahl 0.63 uncoated
2391|1m|carlstahl 0.61 coated

closed 2021-01-29 by julijenv