robotology/mesh-support #1067: Fault button is faulty

gabrielenava, 2021-02-03:
## Brief description of the request/failure

The fault button of iCubGenova04 does not work properly. In particular, it does not stop the robot even after pressing the button. By moving a bit the plug, then starts to work. 

## Detailed context

## Accompanying material

gabrielenava, 2021-02-03:
cc @S-Dafarra

S-Dafarra, 2021-02-03:
It seems similar to https://github.com/robotology/icub-tech-support/issues/467 and https://github.com/robotology/icub-tech-support/issues/670

julijenv, 2021-02-04:
and #1000

julijenv, 2021-02-04:
Hi @gabrielenava , @S-Dafarra , ASA the robot will be ready to start I will check.
I reattach the arms but I could nt launch the robot due to incompaibitla with iRoncub I guess @HosameldinMohamed is checking

violadelbono, 2021-02-05:
Hi @gabrielenava 

We launched the robot multiple times yesterday trying the fault button and it works correctly when the robot is in its standard position. The attachment on the board is strong and bending the cable also didn't affect the functionality of the fault.

Our guess is that the problem is probably somewhere else inside the fault chain of the robot (for instance some cable touching the metal frame) and it could also be due to the ironCub setup (insulating pants, straps and so on).
Unfortunately it is difficult to find the problem in retrospect, so we will check at the time of your next experiment.

S-Dafarra, 2021-02-05:
> Hi @gabrielenava
>
> We launched the robot multiple times yesterday trying the fault button and it works correctly when the robot is in its standard position. The attachment on the board is strong and [...]

Hi @violadelbono @julijenv, as you may know, we soon have a very important demo outdoor. It is very important for us that these sorts of things (this issue and #1000) do not appear at all. Is there something else it is possible to check? Is it possible that some board is to be changed? 

I would kindly ask you to assume that the problem is still there, just to be sure that this is not going to happen when the robot will be without cables on an elevated platform in front of cameras.

julijenv, 2021-02-15:
As the board has been changed BCB (the one that was found dying on the issue, see below AND which is BTW the one on which the FAULT button jack cable is attached to) it would be nice to understand if now the fault is working better, let us know

julijenv, 2021-02-18:
Could you please confirm that the issue is recsolved by havoing changed the BCB board , thx in advance, too much issue opened...
@kouroshD , @GiulioRomualdi , @S-Dafarra  etc...

violadelbono, 2021-03-12:
@gabrielenava what is the status of this issue?

gabrielenava, 2021-03-12:
I didn't work on the robot after opening the issue, maybe @S-Dafarra can give more info than me on the status of the issue

S-Dafarra, 2021-03-12:
> I didn't work on the robot after opening the issue, maybe @S-Dafarra can give more info than me on the status of the issue

I think it was working so far. I don't know if it was a problem with the jack maybe, it seemed a bit bent.

closed 2021-03-12 by violadelbono