robotology/mesh-support #1069: iCubErzelli01: Left arm mid finger loose

ddetommaso, 2021-02-03:
## Brief description of the request/failure
We realized that the middle finger of the left arm is getting loose.

julijenv, 2021-02-04:
Hi @ddetommaso ,
the proximal joint was broken , In my attempt to repair it, I noticed that one of the nut screw was block inside the first phalnge due to an excess of glue. I for the moment succeed to make the joint working again (quick fix) but unfortunately we will need to take the forearm to succeed in either replace the piece or make it work the tensing ssystem again as it was plan to.

At the moment it has been taken from the warehouse

IITCODE |QTy| Desciption
---|---|---
1815|1m|carlstahl 0.63mm

julijenv, 2021-02-05:
Ths to the help of the MWS we successfully drill the nut screw and therefore I replace the nut screw M1.6 by a M2,, I remove all the excess of glue. It is now perfect, I didnt remove the quick fix as the finger behave correctly, next time, It will be enough to rewire as it should be and everything will be alright.

closed 2021-02-05 by julijenv