robotology/mesh-support #1070: [iCubGenova09] Left ankle roll smokes

GiulioRomualdi, 2021-02-04:
## Brief description of the request/failure

Today I was trying to collect some data on the iCubGenova09. To collect the dataset, it was required that the left foot sole was fixed to the ground while the right foot was exerting a force on a compliant environment. In order to enforce this condition as much as possible, I blocked the left foot using my own weight. While executing the actual experiment, I suddenly heard a sparkling noise coming from the left foot and some smoke started to flow near the foot itself. Of course, I immediately cut out the power to all the motors from the main line for safety reasons. The temperature of the external cover of the motor was clearly high as I could feel it by just approaching it with my hand.

Remarkably, the friction on the left ankle roll decreased considerably as you can see in the following video.

https://user-images.githubusercontent.com/16744101/106941831-b13ccc00-6723-11eb-9e1b-29d59ebb4836.mp4

After a brief discussion with @DanielePucci, I decided to start it again to check whether the motor was still alive. And it was.

S-Dafarra, 2021-02-04:
Oh wow. Was it still able to move?

GiulioRomualdi, 2021-02-04:
> Oh wow. Was it still able to move? Have you tried to walk again?

From Motorgui yes. Tomorrow I try the walking 

> Did you get any log? I am wondering if there was any quadrature encoder error

I don't the logger since it was saturated :cry:

DanielePucci, 2021-02-04:
To be noted in @GiulioRomualdi first comment ⚠️ 
> Remarkably, the friction on the left ankle roll decreased considerably as you can see in the following video.

For sure, useful to @maggia80 @pattacini @MrAndrea @fiorisi @marcellosavoldi

MrAndrea, 2021-02-05:
From what I understand from the comments, I imagine 3 possible situations, from grease smoke due to high temperature, or some almost melted cable insulations, or stressed connectors and crimps due to the high current. I have to check the motor and the wiring to understand better.

isorrentino, 2021-02-05:
The walking is still working fine!! :heart:

pattacini, 2021-02-05:
We were arguing in team FIX whether we shouldn't introduce 🔥 somehow in the procedure to purposely reduce friction.

MrAndrea, 2021-02-05:
I just checked the motor together with @isorrentino and @GiulioRomualdi and I exclude electrical problem. It seems a grease problem in the harmonic drive related to the high temperature of the motor forced in holded position for minutes (the power in this condition is about 2-3 hundreds watt and the temperature increase a lot).
I suggest to open the motor assembly to verify the grease status.

MrAndrea, 2021-02-05:
BTW, I also noticed some discrepancy in the motor current reading with the current displayed by the power supply. It might be useful to do a quick check.

fiorisi, 2021-02-05:
> For sure, useful to @maggia80 @pattacini @MrAndrea @fiorisi @marcellosavoldi

If the temperature melted the grease, probably now the joint is more lubricated. This means hat we should check the grease in the other joints. 
Or follow the Team FIX suggestion 🙂

MrAndrea, 2021-02-05:
@fiorisi the joint has lower friction and the smell looks like cooked grease.

GiulioRomualdi, 2021-02-06:
We noticed that the friction on the left ankle roll is reduced and now the current required to move the joint while walking is higher that before.

Yesterday we discussed with @fiorisi and a possible action is to check the status of both the ankles (gearboxes and grease). However I'm not sure if icub-tech plans this activity.

Cc @DanielePucci

pattacini, 2021-02-07:
>However I'm not sure if icub-tech plans this activity.

Rest assured that we will be certainly looking at it.
When and how is a matter of internal planning.

pattacini, 2021-02-07:
>To collect the dataset, it was required that the left foot sole was fixed to the ground while the right foot was exerting a force on a compliant environment. In order to enforce this condition as much as possible, I blocked the left foot using my own weight

By the way, I'm probably wrong but, sticking to such a description, this experiment itself looks suspectly disruptive per se.

pattacini, 2021-02-08:
@salvi-mattia told us that after a quick chat with @fiorisi and @S-Dafarra the opinion is that we should continue working with the robot as it is in order to collect further information and be able to better narrow down possible problems related to this.

S-Dafarra, 2021-02-11:
You can refer to https://github.com/icub-tech-iit/task-force-miscellanea/issues/69#issuecomment-777388897 for some walking datasets. From a first inspection, the current on the left ankle roll seems a little higher compared to the right one, but not excessively higher.

pattacini, 2021-03-12:
We have conducted deep mech maintenance of the robot as per https://github.com/robotology/icub-tech-support/issues/1077#issuecomment-797395221, which may have an impact also on this issue.

pattacini, 2021-03-16:
Given the comments in the [attached report][1], I'll close this issue to use #1077 for further discussions.

Feel free to open it up again if requried.

[1]: https://github.com/robotology/icub-tech-support/issues/1077#issuecomment-797395221

closed 2021-03-16 by pattacini