robotology/mesh-support #1072: 2FOC quadrature encoder phase broken saturates immediately the logger

GiulioRomualdi, 2021-02-06:
## Brief description of the request/failure
Every time we (in this week @isorrentino and I) start iCubGenova09 the quadrature encoder error arises and it saturates immediately the logger. 

This is the consequence of https://github.com/robotology/icub-firmware/pull/159 however right now we cannot  debug anymore the problems that we have with the robot (e.g. #1070)

Possible solutions:
1. Print only once the error (workaround)
2. Understand why there is the error and fix it once for all 

This problem is there since October (please check https://github.com/robotology/icub-tech-support/issues/1004 and https://github.com/robotology/icub-tech-support/issues/1055) 

## Detailed context

This is the error on the left leg. However, it happens also on the right leg and shoulders.
```
 90930,305163 <ERROR>  from BOARD 10.0.1.7 (left_leg-eb7-j2_3), src LOCAL, adr 0, time 143s 803m 884u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + 
```

## Accompanying material
[log_icub-head_yarprobotinterface_1583.txt](https://github.com/robotology/icub-tech-support/files/5936669/log_icub-head_yarprobotinterface_1583.txt)

julijenv, 2021-02-08:
Ciao @GiulioRomualdi  and @isorrentino ,
can you double check that the left leg board EMS7 - 10.0.1.7 has the dsame version of the firmware as the rest of the robot.
It is strange that the robot can move if the message still continue to be printed.
The log file say that it is the left knee which has a problem.
let me know

pattacini, 2021-02-08:
Hi @julijenv 

As per https://github.com/robotology/icub-firmware/pull/159, DIC explicitly required us to disable putting the joint in HF in order only to raise the error message.

pattacini, 2021-02-08:
Hi @ale-git 

While we will be debugging this hardware-wise, do you think it'd be worth investigating it also software-wise?
DIC has been getting this error since long and it's still there despite several cleanups of the sensor.

Any chance there's something weird in the logic? Maybe you could also drop here pointers to the FW responsible for such an handling just for a double check?

pattacini, 2021-02-08:
@julijenv reported such a comment:

>Hi guys,
the only thing that come to my mind about this weird behaviour is that it would be nice to check every single BLESS motor to test accurately the drift with the reading of the optical disk. Probably by doing so, we could clean with more precision the disks and perhaps the error got removed.
cc @S-Dafarra

pattacini, 2021-02-08:
Essentially, @julijenv, you would like to perform on iCub 3 the standard encoders' tests you've been doing in PROTO when delivering a brand new iCub.

What do we need for doing that?
Do we need to adapt some conf files?

S-Dafarra, 2021-02-08:
Btw, I think that this issue is a duplicate of https://github.com/icub-tech-iit/task-force-miscellanea/issues/71

pattacini, 2021-02-08:
>Btw, I think that this issue is a duplicate of (or at least very related to) icub-tech-iit/task-force-miscellanea#71

Actually, I would keep this issue to track the activities on the hardware, whereas the other one is for the software request.

davidetome, 2021-03-16:
## Tests on Optical Encoders Drift
After fixing the problems on the legs I ran successfully the tests to check if the optical encoders are clean.
The `joint 3` (knee) of the left leg and the `jpint 1` (roll) of the right leg keep spotting the following error despite the joints calibrate and move correctly : 
```code
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j2_3), src LOCAL, adr 0, time 2378s 374m 569u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```
```code
[ERROR]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 597s 373m 969u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```
## Left Leg
### j0
[image]
### j1
[image]
### j2
[image]
### j3
[image]
### j4
[image]
### j5
[image]

## Right Leg
### j0
[image]
### j1
[image]
### j2
[image]
### j3
[image]
### j4
[image]
### j5
[image]

## Test data
To plot the following data use :
```code
gnuplot -e " unset key; plot for [col=6:6] 'encDrift_plot_right_leg.txt' using col with lines " -persist
```
es. [col=1:6] plots all the joints in one figure

[encDrift_plot_left_leg.txt](https://github.com/robotology/icub-tech-support/files/6149757/encDrift_plot_left_leg.txt)
[encDrift_plot_right_leg.txt](https://github.com/robotology/icub-tech-support/files/6149758/encDrift_plot_right_leg.txt)

cc @pattacini  @salvi-mattia

pattacini, 2021-03-16:
Thanks heaps @davidetome 👍🏻 

Essentially, despite we had a couple of warnings on dirty encoders, the standard tests verifying whether the encoders may drift were successful, not capable of spotting anything weird in this respect.

We are thus going to clean up the quadrature encoders to remove the warnings outright and prepare the robot to give it to you @S-Dafarra.

At this point, those warnings need to be investigated at a very deeper level. Perhaps, the mechanism/machinery devoted to detect/trigger these conditions may be trapped with false positives, somehow.

davidetome, 2021-03-16:
## Whole Robot calibration ok (run of the entire robot)
I ran the entire robot and all the parts calibrated successfully; maybe you'll need to adjust the legs/ankles playing w/ the Delta according to your needs.

## Robot run
The robot calibrate correctly
https://user-images.githubusercontent.com/6638215/111346355-6e550900-867e-11eb-9244-f33bd804bc2a.mp4

## yarpmotorgui
All joints are ok
[image]

cc @S-Dafarra @pattacini @salvi-mattia @simeonedussoni

salvi-mattia, 2021-03-16:
[image]

pattacini, 2021-03-17:
@ale-git, in the test we performed above we had the following error messages stating that the encoder phase was broken:

```
[ERROR]  from BOARD 10.0.1.7 (left_leg-eb7-j2_3), src LOCAL, adr 0, time 2378s 374m 569u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```
```
[ERROR]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 597s 373m 969u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```

This is certainly not true as the robot was moving just fine and we also logged the profiles of the encoders, which were ok as well.

Do you think there might be false positives in the FW mechanism underlying this check?

pattacini, 2021-03-22:
Thanks @ale-git 

Does what you said include the handling of the `encoder dirty` conditions or is it only limited to the `encoder broken` conditions?

Perhaps, an intermediate step could foresee disabling that check with a configuration option.

ale-git, 2021-03-22:
Yes, I mean all encoder fault detection heuristics. About making them optional, unfortunately this is the most expensive task since it involves the embObjMotionControl software up to the robot configuration parser, so it isn't a very "intermediate step" :)

pattacini, 2021-03-22:
>Yes, I mean all encoder fault detection heuristics

Good to know then, as DIC and FIX have been struggling a "little bit" not to say freaking out w/ these messages 😄

>unfortunately this is the most expensive task since it involves the embObjMotionControl software up to the robot configuration parser, so it isn't a very "intermediate step" :)

I think we can survive w/ the PR we made to silent the messages after a given number. We'll need to address this problem systematically at a certain point.

cc @S-Dafarra @GiulioRomualdi @davidetome @simeonedussoni @salvi-mattia @AlexAntn

pattacini, 2021-03-22:
@S-Dafarra if you agree, I'd close this in favor of https://github.com/robotology/icub-firmware/issues/175.

S-Dafarra, 2021-03-22:
> @S-Dafarra if you agree, I'd close this in favor of [robotology/icub-firmware#175](https://github.com/robotology/icub-firmware/issues/175).

Agreed!

closed 2021-03-22 by pattacini