robotology/mesh-support #1073: Changing cable of thumb - iCubLisboa01

plinioMoreno, 2021-02-08:
## Brief description of the request/failure
Our new member of the team @wevmacedo changed the cable of the thumb that controls the upper limbs. In the process, the magnet got rotated and the reference for the hall effect sensor changed.

@wevmacedo managed to change the cable, but when the robot was restarted the cable broke again, and we think the movement of the magnet caused the cable breaking again. We are sending the picture with a mark on the magnet that rotated.

Do you have a pointer on the instructions on how to calibrate the reference of the magnet?

## Detailed context

## Accompanying material

[image]

julijenv, 2021-02-09:
Hi @plinioMoreno ,
could you please just rewire the cable and relaunch the robot with FAULT button pressed.
after noticing on the yarprobotinterface - messages such as - "yarprobotinterface run normally" just open a yarpmotogui interface.
You will see that even if the robot is not enabled, you can retrieve the values of the encoder of any joint. In your case, the distal joint of the thumb (joint 10) .
Try to move the magnet in order to have an **increasing** value for the entire distal finger when you **close** just this joint of the finger. (the distal value of the encoder is the sum of the 2 hall sensor coming from the 2 last magnets). 
You can compare with another distal joint to mimic the behaviour. 
When you finally get the magnet place correctly you may close the yarprobointerface ,and restart everything without the FAULT button pressed.
let me know

plinioMoreno, 2021-02-11:
Thank you @julijenv for the explanation.

We follow the instructions and we were able to start the robot. However, we have an offset on the joint encoder (joint 10) of around 10 degrees, so the initial position of the joint 10 is not zero.

We were looking into the documentation [here](https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration/#hands-calibration) and [here](http://wiki.icub.org/wiki/Calibration_types_available_on_iCub_and_R1_robots) and our calibrator files are old, because we have only three calibration parameters in this [file](https://github.com/robotology/robots-configuration/blob/1856a02a2eaa26dfae7719f8d09a02d5bba9e6d1/iCubLisboa01/calibrators/left_hand_calib.xml#L16).

We have also an excel file that does some computations to find the correct values for the calibration 
[iCub_Calibration_V1_3_1_iCubLisboa01.tar.gz](https://github.com/robotology/icub-tech-support/files/5965291/iCub_Calibration_V1_3_1_iCubLisboa01.tar.gz)

Let us know how to proceed.

Best regards,

Plinio

plinioMoreno, 2021-02-15:
Hi @julijenv, a reminder about our questions.
Best regards,
Plinio 
> Thank you @julijenv for the explanation.
>
> We follow the instructions and we were able to start the robot. However, we have an offset on the joint encoder (joint 10) of around 10 degrees, so the [...]

julijenv, 2021-02-18:
Hi @plinioMoreno ,
sorry for this late reply, we are a bit in a crazy time here in genova.
Sorry for asking because I'm a bit confused or I cant really understand well from what you wrote.
Is the finger stretched with value of 10 OR a bit closed with a value of 10?
#### **CASE 1:** you need to recalibrate using your xls file.
### CALIBRATION PROCEDURE for CAN-based Robot
- you need to perfectly know from where the config files are taken when the robot is launched (you can retrieve the location by looking at the first lines of the log of yarprobotinterface where you can see **file found**.
- modify the file general.xml
from
[image]
to
[image]
this allows you to see the raw value of the encoder.
- PUSH the red Button in order to FAULT the robot and prevent it from moving.
- TURN ON the motor switch
- wait 45 seconds, in the meanwhile open the xls files (iCub_CAlibration_V1_3_1_iCubLisoboa01.xls)
- launch the yrprobotinterface from a new window shell (on pc104 shell) from the repository from where usually the robot gets its config files when you use yarpmanager. (repository from where you have changed the general.xml file BTW)
- you will see that the log file stop writing and asking you for making sure the RED button is pushed (which should be the case ) and press enter - you may need to press enter quite a few times (up to 10 times, btw every time you are asked for)
- at some point finally, you may read robot run happily
- you can start yarpmotorgui now
- read the value of the distal joint of the thumb when opened and closed
- write down the values inside Vmin and Vmax as explained into https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration/#hands-calibration
- Check on the bottom left corner of the xls file you will see a different tab - click on **LEFT HAND**
- you will now copy the calibration of the left thumb from the green part of the left-hand TAB. 
[image]
into the file iCubLisboa01/calibrators/left_hand_calib.mxl
[image]
- and you will take the white part of the starting part of the left_hadn tab or the thumb ditsal
[image]
and copy it into the file iCubLisboas01/hardware/motorControl/icub_left_hand.mxl
[image]

WHEN DONE:
- press "control+C" 2-3 times to kill yarprobotinterface
- shut down the motor switch.
- yarp clean --timeout 0.5, 2 times and check that NO remaining pending port are open - the one with prefix icub
- return the file general.xml as before the calibration
- release the red button fault
- TURN ON the motor switch
- NOW you are free to start the robot as normal

#### **CASE 2:** you wil need to remove the magnet to make sure the finger got decreasing value when getting to stretched posiion. and later RECALIBRATING as explained above

julijenv, 2021-02-22:
Hi @plinioMoreno ,
did you read this?
does it  solve your problem?
let me know

plinioMoreno, 2021-02-22:
Hi @julijenv,

Yes, we read the instructions, and we should be in the case 1 you explained. I hope @wevmacedo will be able to run the calibration during this week and we will let you know.

plinioMoreno, 2021-02-25:
Hi @julijenv,

Here we got confused a bit looking at the calibration instructions [here](https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration/#calibration-types). It seems that for the newer robots the values `Vmin` and `Vmax` go up until 255, but in our case the values are different. Basically, we just want to confirm that `Vmin` and `Vmax` can have other values and the instructions [here](https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration/#calibration-types) are for other type of robot.

cc @wevmacedo

wevmacedo, 2021-02-26:
Hi @julijenv, and @plinioMoreno.

We have made progress with the calibration. Following the steps mentioned above. 
But somehow, in the first change of the cable from joint 11 Index Proximal (with the spacers and as tight as possible) After a few sections of testing, a gap appears in the motor pulley (the cable gap in the image), and I still haven't figured out how this is possible, or what I'm doing wrong. Where does this "extra cable" come from?

[image]

violadelbono, 2021-03-03:
Hi @wevmacedo,

First make sure that the extra cable comes from the new cable you replaced and it's not a leftover from the repairing. 
Is the finger working also with this gap? It is possible that you did not tight it enough during the wiring. In that case I suggest you to re-cable following the procedure on our [website](https://icub-tech-iit.github.io/documentation/icub_tendons/lower_arm_V2/#322-cabling-the-motor-pulley) starting from:

-  Loosen the nut screws at the base of the finger and try to tense the cable from there.
- If you don't manage to do it, cut the knot and remove the cable from the knot site
- Remove gently the cable from its wiring around the pulleys at the finger's base
- No need to rewire the motor pulley, just tight the cable well from here (finger's base).
- Follow the cabling procedure again from the base of the finger

In case you see the cable is too short because of cutting the knot or you can't eliminate the gap, unfortunately you may need to perform the whole cabling procedure.

wevmacedo, 2021-03-10:
hi @violadelbono,
After I managed to pull the excess cable by the base of the finger and redoing the knot, it ended up solving the cable problem.
With some tests the iCub seems to be perfectly calibrated and functional, thank you all.

closed 2021-03-10 by violadelbono