robotology/mesh-support #1074: The robot shuts down on its own from time to time

S-Dafarra, 2021-02-09:
## Brief description of the request/failure

During some tests yesterday and today the robot shut down on its own. In both cases, the robot was connected with the cable and it was still on the pole when it happened. The LEDS after appear red and **not** blinking. The first time the cable was strongly connected. We had to use a wrench in order to unscrew it.

## Detailed context
I am attaching here also a video after it happened in case there might be some specific LED configuration used for debug.

## Accompanying material

https://user-images.githubusercontent.com/18591940/107362678-5adfdc80-6ad9-11eb-921a-6b958a0928c2.mp4

cc @kouroshD @DanielePucci

julijenv, 2021-02-09:
cc @MrAndrea

S-Dafarra, 2021-02-09:
Thinking about it, something happened also when running with the battery. This morning, we were testing some of the robot demos with Wi-Fi and battery. By mistake, we pressed the fault. Nothing special so far. We first checked the battery level by pressing the button and it was showing a red and a green led (so it should have been between 25% and 50% of charged remaining, we used it for 35 mins). We then restarted the motors with the robot hanging. After about 30secs the robot shut down.

Clearly, the battery levels may not be very precise and maybe restarting the robot, the sudden request of current may have emptied the battery. The weird thing is that the battery was completely discharged afterward. Neither the red LED was lighting up.

My guess would be that in all these cases there is a sort of short circuit that suddenly causes a burst of current. When connected to the cable, the power supply simply stops providing power for a fraction of a second and then restarts. As a consequence, both LEDs appear to be red afterward. When connected via battery, there are not these limitations and the battery is drained because of the burst of current.

Also, considering that the LEDs do not blink I would suppose that the problem is in the circuitry between the power supply and the mechanism that checks the overcurrents.

maggia80, 2021-02-10:
@julijenv @violadelbono , we checked yesterday with @MrAndrea and we discover that:

1) The BCB board resets, due to the overtemperature of the DC/DC converter (U2, see pcb layout [here ](https://github.com/icub-tech-iit/electronics-boards-public/blob/master/power-management/backpack-bcb/output/Assembly/BCB.pdf)that generates the 5V in this board. The cause might be internal to the board (some parts of the board consumes more current than expected) or external (issues with the two 5V FAN connected to it. 

You may check the temperature of this component while disconnecting the FANs and see if this is the cause. If this is not the case, we should replace the board.

maggia80, 2021-02-10:
at sec 26-27 you see that the boards switched off.
https://user-images.githubusercontent.com/6987330/107478693-312ac200-6b7a-11eb-8df0-eb53a00d82f0.mp4

MrAndrea, 2021-02-10:
After checking the power supply, which had the right settings, and verifying that after each restart and after a few minutes the robot turned off, the problem was identified in the BCB board. In fact the other boards remained on, but when the BCB was reset, the BDC board was reset via CAN by the BCB with the outputs off.
We therefore identified a possible problem in the 5V regulator (U2), which was very hot and probably goes into thermal shut-down.
Next steps: remove external loads on 5V (the two fans), and then, if necessary, replace the BCB board with a new one, and check the removed one.

julijenv, 2021-02-10:
OK so to sum up, LEt replace the BCB board and try t add some neoprene in between the pin of the BMON.
another question, I saw quite a lot of pin broekn or removed can you just let me know which one need to be replaced?
[image]
the blue circle show some unmounted spring contacts but the red circle show a broken one. Do I need to replace the broekn one?
let me know

S-Dafarra, 2021-02-10:
Hi all, if the fans are attached to the BCB board, can't we just substitute it? 
Checking if that is the problem may require hours now that it is cold. If we can substitute it, we can restart working on the robot to prepare the demo, and in the meanwhile, the other board can be checked more thoroughly. If the fans were indeed the problem, we can still fix that and have a new board. If instead, the board was having some internal components broken, then we did not waste time on the robot.

MrAndrea, 2021-02-10:
@julijenv yes don't mount the two contacts in the blue circle, this is only to prevent strong vibrations from creating a short circuit.
The contact in the red circle and the contact next to it are useless in the dummy battery, they are only for battery charge.

S-Dafarra, 2021-02-10:
@julijenv referred to me that we don't have a spare board in the lab. I don't know if there is another robot we can butcher to get the board replaced :grin:

MrAndrea, 2021-02-10:
I checked the `BCB` board and I found the confirm of the problem on the 5V regulator, although I didn't find any short or something else that can increase the current consumption. The 5V regulator gives the 5V exactly, but it reaches very high temperature together with its output inductor, as shown in the following thermal image where the hot spot HS1 is about 113°C (inductor) and M2 is about 95°C (regulator U2):

[image]

Moreover, more the input voltage increases, more the temperature increases, and if the board is enclosed in the backpack this effect is amplified till the thermal shutdown.

@julijenv gave me a spare BCB board with defective fault feature and I compared the temperature, regulator and inductor stay stable at about 40-45°C.

I just talked with @Fabrizio69 who said that he will give me a new BCB board from the warehouse, so I will program it and mount in the backpack.

MrAndrea, 2021-02-12:
Yesterday I did further analysis on the defective board and I verified that the 5V power domain of the board, and everything connected, worked as expected. The problem seems in the regulator U2 and its related components, that are connected directly to the power input. I found a spare regulator chip and I changed but the problem remained; I couldn't change other components because I don't have spare parts of them.
In the meantime @julijenv connected the spare BCB board that worked fine at my desk, and after powered on it had the same problem of the first BCB.
We took a new board (destined for S / N: 045) and after a test at my desk to verify correct operation we installed it on the robot, connecting one cable at a time to try to find the problem. Everything was ok until we connected the cable coming from the BBA board, at which point the U2 regulator began to overheat and we immediately turned off everything. Fortunately, the BCB board was not damaged.
I checked the cable and the BBA, everything is ok. The problem is in the dummy battery's BMON board, which is bad. Now we can try to replace this and double check everything.

MrAndrea, 2021-02-12:
iCub works again after replacing the `BCB` and `BMON` with new ones and checking that everything is ok.

julijenv, 2021-02-15:
as the fault button is attached thansk to the jack on the BCB board let us know if having changed the board make #1067 solved

closed 2021-02-15 by maggia80