robotology/mesh-support #1075: iCub right thumb cable is broken

kouroshD, 2021-02-09:
## Brief description of the request/failure

The robot left thumb cable is broken during a teleoperarion scenario.

## Detailed context

## Accompanying material

[image]

S-Dafarra, 2021-02-12:
The thumb seems to be fixed, but it is not working well

https://user-images.githubusercontent.com/18591940/107799529-cd60ee80-6d5d-11eb-943a-bf5fb948fbb0.mp4

julijenv, 2021-02-15:
it has been fixed.
it has been taken from the warehouse, the following pieces:
IITCODE |Qta| Description
1815|1m|carlstahl 0.63mm coated

S-Dafarra, 2021-02-15:
It broke again on its own after several startups.