robotology/mesh-support #1076: iCubGenova01 - reporting issues with PC104 startup and/or icub cameras

xEnVrE, 2021-02-11:
## Brief description of the request/failure

After talking f2f with @julijenv, we decided to open this issue to track some strange behaviors that the users of iCubGenova01 noticed in the last few months.

The two facts (not necessarily related) are the following:

- sometimes, even though quite rarely, powering up the PC104 leads to a cumbersome situation where the usual initial noise of the eyelids repeats over time as if the input voltage to the head assembly was going up and down periodically. When this happens the green light on the [CPU power knob](https://user-images.githubusercontent.com/6014499/107623686-00ba5500-6c5a-11eb-8c44-c2cc4f65dc9a.png) flashes synchronously. It seems like an abnormal current absorption (due to a short circuit) which prevents the normal startup and leads to consecutive reboots. In this situation we normally put the CPU knob to OFF and then try again and everything goes normally. Last event: 11/02/2021

- when running iCub cameras using the `yarpdev` (running on the PC104), it happens that the stream of images suddenly stops. The `yarpdev` does not stop running (i.e. no crashes). In this situation, the `yarpdev` cannot be closed normally from the `yarpmanager` and needs to be killed. If one tries to restart the `yarpdev`, it starts normally but no images are streamed. In this situation the `yarprobotinterface` starts normally.  Last event: 11/02/2021

xEnVrE, 2021-02-16:
I think that the first problem is due some failure in the PSU of the PC104.

This morning after switching on the PC104 the head turned off after some seconds and the light in the power knob was off. I switched off the PSU because I though of a short circuit. After that, I tried turning on the PSU alone (without turning on the PC104) and this was the result:

https://user-images.githubusercontent.com/6014499/108045744-1ce83880-7044-11eb-9961-18f168a1735a.mp4

As can be seen, the output voltage is 0 instead of 13.8 as usual.

@julijenv

julijenv, 2021-02-18:
@maggia80 ,
it seems that the main power supply of the pc104 (but let's 12V as it also powers the MC4 boards) is flaky, it may appear that it is high time for it to be changed.
As a reminder we already have this kind of power supply in the warehouse - icub setup (meaning no need to buy another one)

xEnVrE, 2021-03-10:
Ultimately, the problem is happening even more often which makes difficult to use the robot reliably. Sometimes, even if the power supply starts correctly, after some time the voltage drops which results in a reboot of the PC104.

Can I open a ticket on https://github.com/icub-tech-iit/tickets to start reasoning on the substitution of the PSU?

cc @lornat75 @pattacini

pattacini, 2021-03-10:
I think it's worth it.

vtikha, 2021-03-10:
Yes,it is not usable at the moment. Can no longer turn it on. And when it does it cuts the power randomly.

violadelbono, 2021-03-12:
Hi @xEnVrE,

The PSU has been replaced. The next time you turn the robot on let us know if everything works.

xEnVrE, 2021-03-12:
Hi @violadelbono,

I will for sure today and let you know!

xEnVrE, 2021-03-15:
@violadelbono the robot is working as expected after the change to the PSU.

violadelbono, 2021-03-15:
I close it for now. Open a new one if you will still have issues with the cameras.

closed 2021-03-15 by violadelbono