robotology/mesh-support #1077: Weird noises when moving the right knee and the right hip roll

S-Dafarra, 2021-02-11:
## Brief description of the request/failure
With iCub3 there are weird noises when moving the knee via the motorgui. They are audible also at startup. When moving the right hip roll by hand instead, sometimes it produces a metallic noise that seems to be coming from the hip pitch assembly.

We would need to discuss this with @pattacini @salvi-mattia and @maggia80.

cc @DanielePucci @isorrentino @GiulioRomualdi

pattacini, 2021-02-11:
Thanks @S-Dafarra for reporting 👍🏻 
We will be dealing with this "huge" issue after an internal discussion on how to allocate resources.

pattacini, 2021-02-16:
After today's meeting, it came out that we would need quite some days of downtime in order to carefully inspect all the parts and at the same time install the battery. Keeping some margins, **one workweek** (probably less) should be sufficient.

Our proposal is to start off this maintenance the **first week of March 01-05/03** (we cannot do that before, unfortunately).
Does it work for you?

S-Dafarra, 2021-02-16:
> After today's meeting, it came out that we would need quite some days of downtime in order to carefully inspect all the parts and at the same time install the battery. Keeping some margins, **one [...]

Thanks @pattacini, I think it should be ok. In any case, let me ping other people involved with the experiments to make sure we don't clash with any other deadline: @kouroshD @prashanthr05 @GiulioRomualdi @isorrentino

prashanthr05, 2021-02-16:
> Thanks @pattacini, I think it should be ok. In any case, let me ping other people involved with the experiments to make sure we don't clash with any other deadline: @kouroshD @prashanthr05 [...]

No problem for me. Thanks @S-Dafarra.

pattacini, 2021-03-02:
Today we have begun the maintenance of iCub 3, as agreed.
However, and as we have communicated to @S-Dafarra, the required downtime will extend to the next week.

pattacini, 2021-03-12:
@S-Dafarra please find attached the report of the mech activities, as promised.

[iCub 3 – Deep HW maintenance.pptx](https://github.com/robotology/icub-tech-support/files/6129287/iCub.3.Deep.HW.maintenance.pptx)

When the robot will be back operational, we could check if this issue can be marked as solved or not.

S-Dafarra, 2021-03-12:
> @S-Dafarra please find attached the report of the mech activities, as promised.
>
> [iCub 3 – Deep HW [...]

Thanks a lot!

pattacini, 2021-03-15:
@S-Dafarra a quick update on this.

We've been experiencing a bunch of problems while checking the calibration of the iCub 3. Essentially, due to the quadrature encoders that got dirty during the remounting, unfortunately (although we paid attention), we had the following shaking on 3 joints:

https://user-images.githubusercontent.com/3738070/111164787-ad139200-859e-11eb-8086-20543f9a4bbb.mp4

https://user-images.githubusercontent.com/3738070/111164931-ca486080-859e-11eb-992c-3844bcaa3824.mp4

https://user-images.githubusercontent.com/3738070/111164695-99682b80-859e-11eb-9339-8f90014107c2.mp4

👉🏻 We managed to solve the first two problems as per the following video

https://user-images.githubusercontent.com/3738070/111165188-0a0f4800-859f-11eb-9486-a4c3e917083d.mp4

The left hip yaw is still missing and to sort it out we'd be required to unwire the boards again. As it can be easily figured out, this will take some time. Today and tomorrow, we will thus continue working on this.

pattacini, 2021-03-15:
cc @salvi-mattia @davidetome @simeonedussoni

S-Dafarra, 2021-03-15:
Thanks for the update @pattacini! The right hip roll keeps being a problematic joint :grin: Did you have to tune the joints? Maybe the previous set of gains is no more valid. Back in time we had to use a set of gains on the right that was consistently higher than those on the left:
- right: https://github.com/robotology/robots-configuration/blob/3c71e095a7c1c6b5463518dcbe96746430e588b6/iCubGenova09/hardware/motorControl/right_leg-eb10-j0_1-mc.xml#L50
- left https://github.com/robotology/robots-configuration/blob/3c71e095a7c1c6b5463518dcbe96746430e588b6/iCubGenova09/hardware/motorControl/left_leg-eb6-j0_1-mc.xml#L51

pattacini, 2021-03-15:
Actually, we didn't look at the PID gains (yet); so those are the "old" values you were used to.
Please, @davidetome correct me if I'm wrong.

The remaining shaking is on the left hip yaw, not on the right hip roll and should be due to the dirty (freaking) encoders.

The idea is to finish up as soon as possible the cleaning of this optical encoder and the calibration of the 2 missing magnetic AMO (cc @simeonedussoni), quickly check that all the joints start up correctly, and then decide whether we can keep the robot to perform the quadrature encoders test or give the robot back to you and arrange for such a test one early morning by your system setup (@davidetome).

(Re-)tuning the PID is a task we would like to explore with you guys, in this extent.

S-Dafarra, 2021-03-15:
> (Re-)tuning the PID is a task we would like to explore with you guys, in this extent.

No, no absolutely, I was not asking to retune them. I am totally ok with the old ones. I was just wondering if you needed to change any value.

pattacini, 2021-03-15:
>maybe the "old" gains were too high introducing oscillations.

Yes indeed, that was one possibility we explored together w/ @davidetome. We basically removed the I and halved the P but the shaking was still present. By cleaning up the quadrature encoders, we managed to solve the problem on 2 out of 3 joints.
The last joint is still under maintenance (it's difficult to reach the cleaning hole).

pattacini, 2021-03-16:
🧹 We cleaned up the **left hip yaw** and this sorted out the problem we reported above:

https://user-images.githubusercontent.com/3738070/111329899-c3d5e980-866f-11eb-9d3d-f83902f36c49.mp4

iCub 3 is fully operational now ✔

pattacini, 2021-04-06:
Hi @S-Dafarra 

I know you're still getting along with the iCub 3 after all this long interlude 😀but do you have already any feedback on this regarding the weird noise?

S-Dafarra, 2021-04-07:
> Hi @S-Dafarra
> 
> I know you're still getting along with the iCub 3 after all this long interlude  but do you have already any feedback on this regarding the weird noise?

We managed to run it once on Thursday. We did not notice any weird noise anymore. The only noise we could hear was the normal one related to the belt moving.

pattacini, 2021-04-07:
Thanks for the feedback @S-Dafarra 👍🏻 
Closing then, feel free to open it up again if required.

closed 2021-04-07 by pattacini