robotology/mesh-support #1083: Face expression not working (again)

S-Dafarra, 2021-02-16:
## Brief description of the request/failure
The robot's face expressions stopped working again. They usually start one time over three. It also seems that plugging and unplugging the connector from the face mask does not seem to work anymore.

This is a follow-up of https://github.com/robotology/icub-tech-support/issues/862

## Detailed context

## Accompanying material

DanielePucci, 2021-02-21:
Face expressions did not start multiple times just before today's demo. It dalayed other operations and increased the overall stress in a very bad moment (just before starting)

@julijenv @maggia80

julijenv, 2021-02-22:
As suggested during the D&G event, it would be nice to try to do this withour having to open up every single time the head to find out whats going wrong becaseu the error message actually talk about: "dtr/rts error"
probably when the head start something could happen in a good way or in another (rising front of power line on the face expression on usb link with head or converter UDB to tty - all the cables have been checked and recheck several time, last time by despair we recrimp everything buut as the error return, perhaps and because it has nt been too much retested, let's try this command line:

`eject /dev/ttyAMC0; sleep 1; eject -t /dev/ttyAMC0`

S-Dafarra, 2021-02-23:
> let's try this command line:
> 
> `eject /dev/ttyAMC0; sleep 1; eject -t /dev/ttyAMC0`

It outputs

```
icub@icub-head:~$ sudo eject /dev/ttyACM0
not an sg device, or old sg driver
eject: unable to eject, last error: Inappropriate ioctl for device
```

S-Dafarra, 2021-03-01:
@vtikha noticed that after starting the ``yarpdev`` and the ``emotioninterface``, it is better to wait a while (~30sec) before connecting. This seemed to work during the demo. It may have something to do with the time between you launch the device and the moment at which you start using it.

We may further check this.

S-Dafarra, 2021-03-04:
> @vtikha noticed that after starting the `yarpdev` and the `emotioninterface`, it is better to wait a while (~30sec) before connecting. This seemed to work during the demo. It may have something to [...]

I also waited minutes before connecting, but still I had no luck. I think we were only lucky when the "waiting before connecting" strategy worked.

vtikha, 2021-03-17:
Ok, I confirm that the waiting was irrelevant. This happened again during a recent demo...only way to make it works was to shutdown the head a few times..This still needs to be investigated

S-Dafarra, 2021-06-14:
This is still happening. I tried a few days ago. Initially, it seemed that they did not work, then I sent manually some commands through the emotions raw port to command the eyelids, and then they started working. I am not sure if this was just a case.

DanielePucci, 2021-06-15:
CC @maggia80

DanielePucci, 2021-08-30:
Hi all, this problem keeps occurring and slowing down the demo launch (just today): in stressing moments, it may be a blocking point since it requires a certain number of actions to make them work (i.e. shut down the robot, open the robot head, disconnect the power cable, reconnect the power cable, close the head, and launch the robot interface again). 

Can we take some actions to deal with this issue? 

This is going to be pivotal for the looming demos we have in September and October 

Thanks again

CC
@maggia80 @Uboldi80 @Fabrizio69

Uboldi80, 2021-08-31:
Yesterday,  with @Nicogene (whom i thank) I verified that there was no software problem in the module FACE EXPRESSION.
the software was ok.
after this test I tested the hardware part using the Nancy01 mask which was perfectly functional.
to solve the problem, this morning i replaced the board 736 in the face.
I have tested the face expression trough yarpmanager and work well.

DanielePucci, 2021-08-31:
Thank you very much @Uboldi80 and @Nicogene, definitely a step forward for us when launching demos

S-Dafarra, 2021-09-21:
Unfortunately, this morning, right before a demo, the face expressions failed to start again.

DanielePucci, 2021-09-25:
Unfortunately, this happened also during https://github.com/ami-iit/lab-events-demos/issues/186

Uboldi80, 2021-11-25:
About this problem, we dicuss with @Fabrizio69 to verify during the next swap from ironCub to iCub if is possible to introduce the RFE in the head.
and the i suggest a feasibility study during the swap.
I would to ask to @emilianob80 @gandoo end @GiacomoVassallo13 to help me to verify the situation.

S-Dafarra, 2022-01-31:
Interestingly, we noticed on Friday 28th Jan the same problem also on iCubErzelli02.

Uboldi80, 2022-02-08:
let me know what you want to do about the possible head upgrade.

S-Dafarra, 2022-02-08:
> let me know what you want to do about the possible head upgrade.

From my side, I was simply pointing out that this problem seems to affect not just iCubGenova04, but also other robots with a similar face.

In case case, using the RFE should fix the problem on iCubGenova04. For what concerns iCubErzelli02 I would leave to choice to @vtikha

Uboldi80, 2022-05-05:
I replaced the card that manages facial expressions with one reprogrammed by @sgiraz 
I ask you to give me feedback on the operation.

Uboldi80, 2022-05-09:
unfortunately the reprogrammed card did not work as hoped.
@sgiraz is again working to fix the problem.

closed 2023-02-21 by github-actions[bot]