robotology/mesh-support #1085: [iCubErzelli02] Right distal thumb cable broken

vtikha, 2021-02-17:
## Brief description of the request/failure
The right distal thumb cable broke just by moving the `robotmotorgui` at slow speed. This was most probably due to cable wear and tear as did not go into contact with anything 

## Detailed context

## Accompanying material

violadelbono, 2021-03-01:
Ciao @vtikha,

The right proximal thumb and the right distal index have been repaired:

| IITCODE  |  ALIAS | DESCRIPTION  | QUANTITY  |  
|---|---|---|---|
|2391 | U7194561 | Stainless steel micro-cable, cable construction 7x19mm, nom. diameter 0.45mm, outer diameter 0.61mm, Polyamide coated, AISI 316 - CARL STAHL GMBH| 40cm |
|1815 | CG077063 | Stainless steel micro-cable, cable construction 7x7mm, nom. diameter 0,63mm, not coated, AISI 316| 1m|

vtikha, 2021-03-01:
Fantastic! thank you

closed 2021-03-01 by violadelbono