robotology/mesh-support #1086: iCubGenova01 - right r_pinky misbehaves

xEnVrE, 2021-02-18:
## Brief description of the request/failure

After calibration, while the right ring stays at zero degrees, the right little finger goes to ~ 90 degrees and the encoder in the `yarpmotorgui` reads ~ 125 degrees for the joint `r_pinky`.

If I try to command zero degrees, nothing happens.

If I try to command the maximum value, first the little goes to zero (unexpected) and then both ring and little closes (expected). After that If I try to command zero degrees, both fingers open but then, after some seconds, the little closes again, ~ 90 degrees, and the encoder reads 125.0 for the joint `r_pinky`.

## Detailed context

The issue has appeared today but the right hand was not used for weeks.

julijenv, 2021-02-18:
it looks like we need to recalibrate the last fingers

xEnVrE, 2021-02-19:
Meanwuile, as discussed together, I disabled calibration on `r_pinky` to avoid braking the cable during robot startup.

violadelbono, 2021-03-29:
Hi @xEnVrE, I was thinking to look at this tomorrow, is it fine for you?

xEnVrE, 2021-03-29:
Hi @violadelbono, yes it is fine with me!

xEnVrE, 2021-03-30:
It is working as expected after @violadelbono intervention.

Thank you!

closed 2021-03-30 by xEnVrE