robotology/mesh-support #1087: Remote assistance to kick-start iCubWaterloo01

pattacini, 2021-02-24:
Let me report hereinafter the report @julijenv wrote up after the assistance we gave yesterday to kick start iCubWaterloo01:

>Hi everyone,
today we met virtually on Teams and check the integrity of the robot:
>1. EYELIDS: for the beige part, here a zip file with a different format for the CAD of the piece in order to produce or modify it.
[rc_iit_026_p_003.zip](https://github.com/icub-tech-iit/fix/files/6030601/rc_iit_026_p_003.zip)
>2. SKIN: relooking back to the wiring (https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub2.7/pdf/iCub2.7_E3.6.1_14838_Logic.pdf) I would recommend you to double-check the continuity only for the turquoise wire from the bottom of the connector on the top board of the knuckles up to the MTB3 board at the level of the wrist. (it is the continuity to the green wire that we checked in between palm and top board on knuckle)
[image]
[image]
>
>Let me know your findings.

cc @amaroyo @kothiga

amaroyo, 2021-03-02:
Hi @pattacini, @julijenv, 

Thanks for the info, I will check the wiring in the next couple of days and let you know. 

Regarding the zip file, I cannot access it (not found)... if it's easier, and the file is not too big, you can send it via email! 

Thanks, 
Alex.

pattacini, 2021-03-02:
Here's the file!

[rc_iit_026_p_003.zip](https://github.com/robotology/icub-tech-support/files/6069537/rc_iit_026_p_003.zip)

Sorry, it was a copy and paste from another private repo. Now it should work.

maggia80, 2021-03-03:
Hi @amaroyo 

I think that the issue of the eyelids could be solved without changing any part, but just pressing a bit on the part shown below:

[image]

The bigger gap could be due to a deformation of the sheet metal part here: 
[image]

If you press the green part by hand you should reduce the gap. Once you have done it you need to calibrate it, since the magnet changed its initial position.

cc: @violadelbono

amaroyo, 2021-03-04:
Hello, 

@julijenv, #2: Honestly, I couldn't find the turquoise cable... but, it seems to be working now without any issues! Maybe something was slightly loose, or not well attached due to the transportation but now, after having it opened up several times, seems to be ok... I will still keep an eye on it, just in case!

@maggia80, #1: Thanks for the tip! All the parts are quite tighten, but I did press the green part as well. It reduces the space, and it engages the gear of the eyelids, but when released, it goes back to its previous position. 

We are in the process of printing that piece, and we can also compare the sizes - to verify if it has shrunk. If that's not the case, we can of course print one that's a millimeter bigger... but I am concern that there would be a bigger underlying issue - some kind of displacement in the head that happened during the transportation. 

We'll let you know somewhere next week. 

Thanks, 
Alex.

amaroyo, 2021-03-05:
@julijenv, any tip on how to pop the eye lids? Or you'd prefer to set up a quick call to go through the process of changing the beige part? Thanks!

maggia80, 2021-03-08:
hi @amaroyo, @julijenv is not available. However, for what concern the eyelids, I think that what I suggested you it should work. It's not due to a complete displacement, but it could be due to stress in this part during the travel. In order to increase the bending you can apply, you should remove the eyelids, press the part, and put the eyelids back to their position.

maggia80, 2021-03-08:
in the video I'm pressing it gently. If you press a bit more you should gain 1mm or so in order to remove the gap.  
https://user-images.githubusercontent.com/6987330/110309808-15dc9680-8002-11eb-97e5-ceac923f8b69.mp4

amaroyo, 2021-03-15:
Hi @maggia80, 

Just to let you know that we measured the beige part and it didn't shrink. We wanted to replace it with another one that's 1mm bigger, but we couldn't as too many pieces had to be removed in order to access those screws that hold it. 

So, we proceeded with your suggestion; removing the eyelids we managed to force that piece down but the motor touches the right eyelid - playing a bit with it, we managed to put it in a good position, however, all the structure is a bit loose. It is ok for a temporal solution. 

Could you please give me the CAD model for the plastic semi-circular gear piece that sits on top of the eyelids? It seems easier to replace, and I can print it 1mm bigger so the motor won't touch the eyelid. 

Thanks, 
Alex.

violadelbono, 2021-03-16:
Hi @amaroyo,

You can find the CAD of the robots in this repository: https://github.com/icub-tech-iit/cad-mechanics-public
Go into `projects` ->` icub/icub1/icub2 `
I think your piece is `RC_IIT_023_R_003 
`

amaroyo, 2021-03-16:
Hi @violadelbono, 

Do you mind exporting the specific piece? I tried opening up the repository but there are more than 3k pieces and it just gets stuck... 

Thanks, 
Alex.

violadelbono, 2021-03-17:
Here @amaroyo :

[hpc_zpg_0-4_100.prt.zip](https://github.com/robotology/icub-tech-support/files/6154656/hpc_zpg_0-4_100.prt.zip)
[rc_iit_023_r_003.asm.zip](https://github.com/robotology/icub-tech-support/files/6154622/rc_iit_023_r_003.asm.zip)

amaroyo, 2021-03-17:
@violadelbono, awesome, thanks! :) 
I'll try to test it by the end of the week.

pattacini, 2021-03-28:
Closing, feel free to open it up again if needed.

closed 2021-03-28 by pattacini