robotology/mesh-support #1088: Left arm wrist not calibrating correctly

lunagava, 2021-03-01:
## Brief description of the request/failure
The wrist yaw of the left arm (iCubGenova02) seems not calibrating correctly with the fingers pointing slightly upwards (not pointing straight).

## Detailed context

## Accompanying material

lrapetti, 2021-03-04:
Adding here a picture a reference picture I took today.
I think also the right one might not be perfectly calibrated (pointing a bit downwards?)

[image]

violadelbono, 2021-03-10:
Wrist pitch and yaw recalibrated (here the [commit](https://github.com/robotology/robots-configuration/commit/3c71e095a7c1c6b5463518dcbe96746430e588b6)).

closed 2021-03-10 by violadelbono