robotology/mesh-support #1090: Cameras did not start after unpacking the robot

S-Dafarra, 2021-03-03:
## Brief description of the request/failure
During a demo, as soon as we unpacked the robot, the cameras did not start. Here a screenshot of the error when launching the camera device:
[image]

After rebooting the head, the issue did not appear anymore. So this issue is mainly for reporting the problem and discussing about possible causes.

## Detailed context

## Accompanying material

S-Dafarra, 2021-03-03:
This is the line of code that throws the error: https://github.com/robotology/icub-main/blob/6a27bc28c9570db7f96a3e033a563f1ec6291ce3/src/libraries/icubmod/dragonfly2/linux/FirewireCameraDC1394-DR2_2.cpp#L335

violadelbono, 2021-03-29:
Hello @S-Dafarra, I put this issue in the Feedback pipeline, please write here if it happens again.
It could be an isolated event.

closed 2021-06-23 by github-actions[bot]