robotology/mesh-support #1091: The "fake fault" fell into two pieces

S-Dafarra, 2021-03-03:
## Brief description of the request/failure
During a demo where the robot had to walk without any wire, we used the so-called "fake fault". It is just a jack that emulates the presence of the fault without using the button. During this demo, it divided into two pieces. The robot did not fault because luckily the jack remained connected, but that really scared us since the robot was without the crane.

I wonder if we could glue the two pieces together.

cc @DanielePucci 

## Detailed context

## Accompanying material

violadelbono, 2021-03-29:
Hi @S-Dafarra,

If you want to glue the two pieces together I can provide you the glue.

S-Dafarra, 2021-06-14:
I think we can close this

closed 2021-06-14 by S-Dafarra