robotology/mesh-support #1093: Right arm connection problems

S-Dafarra, 2021-03-04:
## Brief description of the request/failure
As soon as we start the robot, the logger is filled with the following error
```
1248.465876	ERROR		 from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src CAN1, adr 0, time 179s 549m 119u: (code 0x00000019, par16 0x0401 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .

```

The robot is able to start, but the right arm stops working after a while.

## Detailed context

## Accompanying material

S-Dafarra, 2021-03-04:
Here a log after a clean start 
[log_icub-head_yarprobotinterface_4308_right_arm_problems.txt](https://github.com/robotology/icub-tech-support/files/6084966/log_icub-head_yarprobotinterface_4308_right_arm_problems.txt)

violadelbono, 2021-03-04:
The board 27 is connected to the arm skin:

[image]

So I think some MTB boards are missing or maybe disconnected. By running Firmwareupdater from icub-head, if you make a discovery under the board eb27, you should see if/which MTBs are missing. Ideally you should see CANID from 8 to 14.
If you manage to do it, let me know.

S-Dafarra, 2021-03-04:
Actually, I see none

https://user-images.githubusercontent.com/18591940/110000411-8c804800-7d13-11eb-96ec-b2fa14a49262.mp4

violadelbono, 2021-03-04:
Could you please check what's under the board eb 24 (the other arm)? 

It is possible that the skin has been disconnected. I will need to check the connections.

S-Dafarra, 2021-03-04:
> Could you please check what's under the board eb 24 (the other arm)?
> 
> It is possible that the skin has been disconnected. I will need to check the connections.

Here:
[image]

violadelbono, 2021-03-04:
Grazie! 
I have the feeling that all the skin of the right arm has been disconnected during/after the demo (on purpose? Idk) so I will check it.

S-Dafarra, 2021-03-04:
> Grazie!
> I have the feeling that all the skin of the right arm has been disconnected during/after the demo (on purpose? Idk) so I will check it.

Not sure about it. It is something that appeared today between one reboot and the other. I tried the robot once as soon as we came back and this error was not there.

violadelbono, 2021-03-08:
The power of the right upper arm skin was disconnected. We can close.

closed 2021-03-08 by violadelbono