robotology/mesh-support #1094: Left arm elbow goes in HF and FT produces spikes

S-Dafarra, 2021-03-05:
## Brief description of the request/failure

During a recent demo, the left elbow was going often in HF. If I remember well, the error was related to the AEA encoder, but unfortunately, I cannot retrieve a usable log at the moment due to #1093 that saturates the logger immediately. 

In addition, from time to time, the FT on the same arm was producing spikes. We did not observe a similar behavior in the other arm, so I suppose there could be a cabling issue.

## Detailed context

## Accompanying material

violadelbono, 2021-03-16:
Today it happened again. With @Gandoo we opened the back of the robot and inspected the boards behind the backpack. The cables of the left elbow AEA sensor were loose and have been re-crimped.

we can have feedback later from @vtikha to know if this solved the problem or not

vtikha, 2021-03-17:
So the intervention that @violadelbono and @Gandoo did was successful. The left arm worked flawlessly after that. 
We used it constantly for 6-8 hours after the intervention.

closed 2021-03-17 by violadelbono