robotology/mesh-support #1096: The point to gaze application may make the neck performing very wide and unstable movements

S-Dafarra, 2021-03-05:
During a recent demo, we tried to use the application that allows to look at the point clicked on the ``yarpview``. In some case, the clicking on the viewer resulted in a very wide motion of the neck, setting the neck joints to hardware fault. We did not manage to find the root of the cause.

cc @vtikha @pattacini @vvasco @DanielePucci

pattacini, 2021-03-05:
Issue transferred here as it has likely very little to do w/ funny-things.
More to come.

pattacini, 2021-03-05:
Probably, there were low-level problems with the neck.

My suggestion is to perform the following tests, simulating the click.

1. Launch `iKinGazeCtrl` and verify which cameras resolution it will be using (just read the printouts)
2. Connect to `iKinGazeCtrl/rpc`
3. Send out streams of commands of the type `look mono (left u v 1.0)` where `u` and `v` are image coordinates within either 320x240 or 640x480 range.
4. Observe the outcome.

In case the weird and unstable behaviors will persist, one could try out to disable the IMU by launching `iKinGazeCtrl --imu::mode off` (and check that IMU gets disabled by inspecting the printouts).

Relevant documentation at https://robotology.github.io/robotology-documentation/doc/html/group__iKinGazeCtrl.html.

vtikha, 2021-03-05:
I ran a few tests in various configurations  when we got back  in order to understand and close in on the cause of this behaviour. 

The system behaved properly even when "extreme" positions were given with fast changes. I checked every point given to `iKinGazeCtrl` and everything was ok.

Double checked all the configuration files + resolutions that we changed with @vvasco for the `cameras` and `iKinGazeCtrl` and they are consistent and correct.

Here is a video of the gazing:
https://user-images.githubusercontent.com/4537987/110115276-25679f80-7db6-11eb-80d6-88f71a35fc49.mov

What I still wanted to check are the following:

1. Attach a secondary monitor to the laptop and use the `yarpview` with that one.
2. Check the switch of **control modes** between` ctpservices` and the `point to click`. It seemed ok but need to make sure.
3. Run the `point to click` just after having the robot walk for a bit to replicate the conditions during the demo (IMU?) 
4. running out of ideas...

Will update as soon as I perform the tests.
More later

pattacini, 2021-03-05:
Great!
Thanks @vtikha

closed 2021-06-23 by github-actions[bot]