robotology/mesh-support #1097: Connecting the robot to the charger may set the 2FOC boards to hardware fault

S-Dafarra, 2021-03-05:
## Brief description of the request/failure
We experienced this during a recent demo. It was happening only with a specific charger (the blue one). When the charger started charging the battery (some seconds after connecting it to the battery, you can hear a relay clicking when it starts), all the joints controlled with a 2FOC (basically all of them except the head and the wrists) were going in HF.

We still need to check if this is still happening after replacing the connector and if this affects only one battery.

cc @maggia80 @MrAndrea 

## Detailed context

## Accompanying material

MrAndrea, 2021-03-05:
I remember this problem at the D&G demo, at the same time we've  had the other black battery charger faulty.
First idea, the two battery charger are different, and it might be useful to check the voltage when the blue one is connected to the battery, to see if there are spikes or something noisy for the power line during the relay toggling.

S-Dafarra, 2021-03-08:
I just tried with the "new battery" (the one with full capacity) when it was at 43% and it did not happen. I don't know if the new prolongue might be helpful.

By the way, I don't know if it is worth improving that prolongue, it seems a bit fragile.

maggia80, 2021-03-09:
> I just tried with the "new battery" (the one with full capacity) when it was at 43% and it did not happen. I don't know if the new prolongue might be helpful.
> 
> By the way, I don't know if it is worth improving that prolongue, it seems a bit fragile.

Yes, we should make it better. we'll take care of that.

emilianob80, 2021-04-14:
@maggia80 Intervention scheduled for Friday 16/04/2021

cc: @S-Dafarra @MrAndrea

emilianob80, 2021-04-16:
see issue https://github.com/icub-tech-iit/proto/issues/996

cc: @violadelbono @maggia80 @S-Dafarra @MrAndrea

S-Dafarra, 2021-09-29:
This is still happening also with the new charger. cc @maggia80 @MrAndrea

MrAndrea, 2022-09-22:
Old issue, no problem has been reported anymore, can we close?

S-Dafarra, 2022-09-22:
We did not report it anymore because we did not use iCubGenova04 with the battery since the last comment. On the other hand, we can also close this and reopen it in case we have further evidences

closed 2022-09-22 by maggia80