robotology/mesh-support #1099: iCubErzelli01: Neck issues

ddetommaso, 2021-03-16:
## Brief description of the request/failure
After around two weeks running the same experiment without any problem, in the past few hours the neck sometime get stuck requesting to look at looking at requests (towards left/right). Attached you can find the logs

## Accompanying material
[log.txt](https://github.com/robotology/icub-tech-support/files/6146824/log.txt)

maggia80, 2021-03-17:
`5384	848.943519	INFO	 from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 900s 93m 218u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> eoerror_INTERNALERROR__valueisempty + .`

@marcoaccame  could you help us on interpreting this INFO?

violadelbono, 2021-03-17:
The connector of the motor of the roll joint was not totally inserted inside its board. It has been also re-crimped to ensure a correct contact.

[image]

Hopefully this solved the problem.

ddetommaso, 2021-03-17:
Everything works great again!

closed 2021-03-17 by violadelbono