robotology/mesh-support #1101: Left wrist cables loose

S-Dafarra, 2021-03-17:
## Brief description of the request/failure
The left wrist seems very loose. I guess that some cable need to be tighten

https://user-images.githubusercontent.com/18591940/111440455-a7d15700-8706-11eb-8619-27c72a83becc.mp4

I can also help in tightening them up if you point me to the correct screw.

## Detailed context

## Accompanying material

violadelbono, 2021-03-17:
@S-Dafarra I believe [these ]( https://icub-tech-iit.github.io/documentation/icub_tendons/wrist_hand_cabling_V2/#13-wrist-passive-tendons)are the tendons which need to be tensed. 
You can try to tight them. You should only need a cross screwdriver but be careful, they are tiny and delicate!

Let me know

[image]
[image]

S-Dafarra, 2021-03-17:
Thanks @violadelbono, I tried to tighten them a bit, but not a lot. I think it improved a bit, at least it is comparable to the other hand. I think that the tendon might be a little long, if that is possible. In any case, I think it is not a big problem and we can close this. If this turns out to be a problem, I will open a new issue.

closed 2021-03-17 by S-Dafarra