robotology/mesh-support #1103: Left hip roll AEA encoder cabling faulty

S-Dafarra, 2021-03-18:
## Brief description of the request/failure
The robot started correctly, and after a while, the encoder on the left hip pitch measured -360 degrees. In the log there are a lot errors like
```
	3183,469360	WARNING		 from BOARD 10.0.1.6 (left_leg-eb6-j0_1), src LOCAL, adr 0, time 2134s 386m 952u: (code 0x0200000a, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
```

## Detailed context

Here the encoder value
[image]

Here the log
[log_icub-head_yarprobotinterface_13760_left_hip_pitch_encoder_broken.txt](https://github.com/robotology/icub-tech-support/files/6163367/log_icub-head_yarprobotinterface_13760_left_hip_pitch_encoder_broken.txt)

## Accompanying material

pattacini, 2021-03-18:
cc @simeonedussoni @davidetome

davidetome, 2021-03-18:
I'm going to check it w/ the support of @emilianob80 after lunch

emilianob80, 2021-03-18:
> I'm going to check it w/ the support of @emilianob80 after lunch
> 
> cc @S-Dafarra @pattacini

the sensor cable is ok

davidetome, 2021-03-18:
I noticed the same behaviour at the first run, the joint was measuring 365 degrees; so I calibrated again the joint and now it seems to work. 👍🏻 
 I ran a sequence for about 10 minutes.
Anyway, keep this under control and if it happens again we should check the AMO sensor w/ @simeonedussoni setup.

@S-Dafarra the calibrator file of the left leg is changed in the robot folder, I did;t install it.

cc @pattacini @emilianob80

S-Dafarra, 2021-03-18:
It happened again the first time I run the robot with @emilianob80. Initially, I thought there was a mismatch in the configuration files, but I double-checked and they were ok.

Then, I rebooted the motors, run from the source dir and it did not happen. Rebooted the motors again and tried again from source, still ok. Then I tried a third time, this time from the ``yarpmanager`` and it was still ok. So it seems to be appearing/disappearing when rebooting the motors.

simeonedussoni, 2021-03-18:
tomorrow morning I will give a look at the AMO encoder to make sure it works properly. 
cc @S-Dafarra @pattacini @davidetome

S-Dafarra, 2021-03-18:
It happened again while I was literally looking at it. It was perfectly still and all of a sudden the encoder jumped to -351. This happened after less than 10 minutes after we did a clean restart for the exact same problem.

simeonedussoni, 2021-03-19:
I re-calibrated the AMO board, it should not need the high-level joint calibration. results from the calibration were pretty ggod so if the problem persists the cause can be elsewhere.

S-Dafarra, 2021-03-22:
> I re-calibrated the AMO board, it should not need the high-level joint calibration. results from the calibration were pretty good so if the problem persists the cause can be elsewhere.

Hi @simeonedussoni did you try to run the robot by any chance?

I tried to run it but I received the error
```
	229,021798	ERROR		 from BOARD 10.0.1.6 (left_leg-eb6-j0_1), src LOCAL, adr 0, time 15s 147m 367u: (code 0x05000016, par16 0x2003 par64 0x0000000000ff0099) -> CFG: EOtheEncodeerReader cannot be configured: verification fails. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
	238,558185	ERROR		 from BOARD 10.0.1.6 (left_leg-eb6-j0_1), src LOCAL, adr 0, time 24s 679m 841u: (code 0x02000008, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
```
In addition, now not even the roll works anymore
[image]

simeonedussoni, 2021-03-22:
We plan a sequence of operations to track the origin of this problem:
1. check values (static and with hand-driven movements) outputted from the yarpmotorgui
2. check the firmware version on board 10.0.1.6
3. check if there can be a wrong memory allocation of the calibration data 
4. check the AMO encoder working (just to avoid unmounting and mounting of the covers withh possible cable breaks etc.)

once (hopefully) fixed, I will go for a complete test run.

davidetome, 2021-03-22:
We tried to do :

-  calibration on Joint 0,1 of the left leg but the reading is stuck to 0, probably a cabling issue.
-  moving joint 3 of the right leg but it stops going in HW FAULT after a while raising the i2T overcurrent error.

cc @simeonedussoni @pattacini @S-Dafarra

simeonedussoni, 2021-03-23:
> calibration on Joint 0,1 of the left leg but the reading is stuck to 0, probably a cabling issue.

joint 0 was slightly unplugged, replugged and calibrated, here the calibration outcome:
[image]

though, the joint is still going to hardware fault, outputting a value of the angle (currently) of $-1081^o$

joint 1 was plugged in the wrong slot, calibration as good as joint 0, going into idle state (probably is it coupled with joint 0 in the control logic?)

simeonedussoni, 2021-03-23:
still struggling with joint 0. it starts measuring correct values then it suddeny subtracts 360 degrees and consequently goes into Hardware fault

simeonedussoni, 2021-03-23:
after modifying and reverting back the multiturn configuration of the `AMO` sensor seems to be solved, at least it ran >10 minutes without errors

here the configuration for the multi-turn option
 
[image]

pattacini, 2021-03-23:
That's good news @simeonedussoni 🚀 
So now both join 0 and 1 are working normally?

S-Dafarra, 2021-03-23:
Nice! So for some reason, the encoder was counting some full turns?

simeonedussoni, 2021-03-23:
> 
> 
> That's good news @simeonedussoni 🚀
> So now both join 0 and 1 are working normally?

yes, now I manage to perform a cycle to have the joints moved for a while (with the help of @davidetome ) in order to make sure the problem is no more reissued

> 
> 
> Nice! So for some reason, the encoder was counting some full turns?

exactly, and the turn number sometimes skipped by one or more (at least, the effect was something like this)

davidetome, 2021-03-23:
# Update

### Right leg
The right leg seems ok now, it cycled for about 15 minutes

https://user-images.githubusercontent.com/6638215/112153668-cb534080-8be3-11eb-84af-68e1c2f66f39.mp4

### Left leg
The left leg can't calibrate the hip roll (joint 1) unless helped by hand; I checked the XML files and I noticed that the Kp of the left leg is very lower than the right one. I tried to put the same value butt nothing changes, The motor-control params are the same.

### cali falis
[image]

### calib ok if helped by hand

https://user-images.githubusercontent.com/6638215/112154264-60eed000-8be4-11eb-8840-8e2e7430b23a.mp4

https://user-images.githubusercontent.com/6638215/112154594-b6c37800-8be4-11eb-819f-641b6d6525d1.mp4

cc @simeonedussoni @pattacini @S-Dafarra @ale-git

simeonedussoni, 2021-03-23:
may be an effect of the mechanical maintenance? is there a methhod to understand how high is the stiffness/frction in a joint, either by analyzing the dynamical behaviour (e.g. velocity vs current curves...) or by doing a mechanical measurement on the joint (e.g. pulling it while measuring the exerted force with something like a dynamometer?)

cc @fiorisi @salvi-mattia

S-Dafarra, 2021-03-23:
> The right leg seems ok now, it cycled for about 15 minutes

Was the cabling of the knee encoder checked, by any chance? I suspect it was not just a temporary problem

> The left leg can't calibrate the hip roll (joint 1) unless helped by hand; I checked the XML files and I noticed that the Kp of the left leg is very lower than the right one. I tried to put the same value butt nothing changes, The motor-control params are the same.

This is new. When we had the sensor working, we did not have these problems. Even after the deep maintenance we did not have issues in that phase. Does it print an error on the quadrature encoder too? Maybe there is a problem in controlling the motor properly.

Otherwise, if ti is just matter of friction, maybe https://github.com/robotology/robots-configuration/blob/3c71e095a7c1c6b5463518dcbe96746430e588b6/iCubGenova09/calibrators/right_leg-calib.xml#L34 can have a role.

pattacini, 2021-03-24:
@davidetome is gonna test w/ an increased PWM limit in calibration.

fiorisi, 2021-03-24:
> may be an effect of the mechanical maintenance? is there a methhod to understand how high is the stiffness/frction in a joint, either by analyzing the dynamical behaviour (e.g. velocity vs current [...]

Moving the left and right joints by hand "feels" similar?

simeonedussoni, 2021-03-24:
can we perform some moving of the joint with a dynamometer to have an estimate of the force to be apllied?

fiorisi, 2021-03-24:
> can we perform some moving of the joint with a dynamometer to have an estimate of the force to be apllied?

That would be more quantitative for sure.
Another test could be performed using the "grey box" to control the brushless motors. Probably is still in the workshop.

ale-git, 2021-03-24:
I agree with @pattacini , it seems a calibration PWM limit issue.

davidetome, 2021-03-24:
> I agree with @pattacini , it seems a calibration PWM limit issue.

@ale-git Yes it is, I set 5000 instead 3000 for the `startupPwm` param in the calibrator file and it works; so I propose to set this param to 5000 for all the joint of the hips.

> Moving the left and right joints by hand "feels" similar?
@fiorisi I tried by hand but is difficult to estimate any differences but moving the joint with a low speed (10) the left one seems less smooth than the right one may be due to a more friction

Finally, the right knee spotted again the problem and doesn't move again. We could check the ROIE sensor cabling or try to clean it again?

cc @pattacini @ale-git @S-Dafarra

pattacini, 2021-03-24:
>I propose to set this param to 5000 for all the joint of the hips.

I'm in favor of this 👍🏻 

Also, some other quantitative analyses can be carried out in the background.

ale-git, 2021-03-24:
5000 is pretty high btw. I have a doubt: have been the motor disassembled lately? I mean: is the stator orientation changed inside its housing? In this case the affected motors need re-calibration, otherwise the torque is not optimal.

[image]

pattacini, 2021-03-24:
>is the stator orientation changed inside its housing?

cc @salvi-mattia @fiorisi

salvi-mattia, 2021-03-24:
no

S-Dafarra, 2021-03-24:
What would that parameter mean? Does it depend on the motor encoder?

pattacini, 2021-03-24:
>What would that parameter mean? Does it depend on the motor encoder?

I was just on a call w/ @ale-git a few minutes ago. 
There's an automatic procedure to let the system find that relative position. Unfortunately, the procedure is not documented.

I would propose to go this way:
- We proceed now w/ 5000 PWM in calibration.
- We document the procedure.
- We run the automatic procedure in background to see whether we'll end up with the same values or not.

Does it sound ok?

davidetome, 2021-03-24:
> 5000 is pretty high btw. I have a doubt: have been the motor disassembled lately? I mean: is the stator orientation changed inside its housing? In this case the affected motors need re-calibration, otherwise the torque is not optimal.
> 
> [image]

For calibration do you mean the encoder reading at the zero position or the params highlighted in the image?

S-Dafarra, 2021-03-24:
> I would propose to go this way:
> 
> * We proceed now w/ 5000 PWM in calibration.
> * We document the procedure.
> * We run the automatic procedure in background to see whether we'll end up with the same values or not.
> 
> Does it sound ok?

Ok for me

ale-git, 2021-03-24:
@davidetome yes, the highlighted parameter. The calibration value is obtained by running once the motor with RotorIndexOffset=1 and AutoCalibration=1. The yarprobotinterface prints messages with the new RotorIndexOffset to be written in the configuration file. It is necessary only if the stator has been moved/removed.

pattacini, 2021-03-24:
Ok, so we need to do this asap then.

davidetome, 2021-03-24:
> Ok, so we need to do this asap then.

I'll do that just after lunch, thanks heaps @ale-git 👍🏻

pattacini, 2021-03-24:
@davidetome I think we should check this way all the motors of the legs.

davidetome, 2021-03-24:
@ale-git I ran `yarprobotinterface` for the left leg with the flags set as you explained and I read the `offset=XXXX` numbers, but putting the values on the respective joints, I got the error : 
```
code
[ERROR] In  left_leg-eb7-j2_3 joint  0 : rotorIndexOffset should be in [0,359] range.
```
Actually the offset values reported are much bigger than 359

This is the log : 
[ll-log.txt](https://github.com/robotology/icub-tech-support/files/6197596/ll-log.txt)

Do I have to set the first two flags in `general.xml` to true as per joints calibration?

ale-git, 2021-03-24:
@davidetome it isn't the right offset, it regards normal startup calibrations. Let me see in the firmware...

davidetome, 2021-03-24:
> @davidetome it isn't the right offset, it regards normal startup calibrations. Let me see in the firmware...

It was the only offset I found 😅

ale-git, 2021-03-24:
@davidetome for which joints did you set the AutoCalibration?

davidetome, 2021-03-24:
> @davidetome for which joints did you set the AutoCalibration?

@ale-git all the joints of the left leg

ale-git, 2021-03-25:
Hi @davidetome, unfortunately the autocalibration message hasn't a dedicated tag in the EMS so it is sent as a generic DEBUG message and it must be interpreted (I believe it should be fixed). In the log file I can find the autocalibration values for three joints: 

J0 = 193
J1 =   12
J3 = 150

Can you try again to autocalibrate the missing others? Maybe you didn't wait enough...

pattacini, 2021-03-25:
>(I believe it should be fixed)

Yep, we'll fix it in background.

davidetome, 2021-03-25:
@ale-git how can I check if the messages are printed for all the joints?

ale-git, 2021-03-25:
This is an autocalibration message:

`from BOARD 10.0.1.7 (left_leg-eb7-j2_3), src CAN1, adr 2, time 142s 360m 958u: (code 0x04000002, par16 0x8129 par64 0xffea0062c7be0096) -> DEBUG: tag02 + .`

looking at par64, the rightmost 4 digits are the offset: 0x0096 = 150 degrees. 8 leftmost digits (here 0xffea0062) must not be 0 or the calibration is failed/not completed. The joint number is determined by its can address, i.e. CAN1, adr 2 ==> J3

davidetome, 2021-03-25:
@simeonedussoni the encoder of the left leg hip pitch (joint 0) is stuck to 0, can you help me checking the wires?

davidetome, 2021-03-25:
> @simeonedussoni the encoder of the left leg hip pitch (joint 0) is stuck to 0, can you help me checking the wires?

@simeonedussoni same for right leg hip roll (joint 1)

davidetome, 2021-03-25:
I found:

 - Left leg : LS0 (AMO joint 0) w/ connector disconnected
 - Right Leg : RS1 (AMO joint 1) the red wire (positive pwr) uncrimped

cc @simeonedussoni

davidetome, 2021-03-26:
@ale-git I tried to run it on the right leg but I can find only that kind of message for joints 0,1,3 as for the left leg

## j0_1
```
tumme@pop-os:~/Documents$ grep j0_1 log.txt 
[DEBUG] eth::parser::print(boardData) for BOARD right_leg-eb10-j0_1
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS/Name =  right_leg-eb10-j0_1
[DEBUG] eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.10 (right_leg-eb10-j0_1) with timeout = 0.02 sec and period of missing report = 60 sec
[WARNING] In  right_leg-eb10-j0_1 joint  0 : motor autocalibration is enabled!!! ATTENTION!!!
[WARNING] In  right_leg-eb10-j0_1 joint  1 : motor autocalibration is enabled!!! ATTENTION!!!
[INFO]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 85s 525m 485u: (code 0x00000034, par16 0x0001 par64 0x0000000000000017) -> SYS: the board has detected a restart of the remote transceiver because the rx ropframe sequence number is 1. In par64 there is the expected number. + .
[DEBUG] EthResource::verifyBoardPresence() found BOARD right_leg-eb10-j0_1 with IP 10.0.1.10 after 0.00319123 seconds
[DEBUG] EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD right_leg-eb10-j0_1 with IP 10.0.1.10
[DEBUG] EthResource::cleanBoardBehaviour() has cleaned the application in BOARD right_leg-eb10-j0_1 with IP 10.0.1.10 : config mode + cleared all its regulars
[DEBUG] EthResource::setTimingOfRunningCycle() for BOARD right_leg-eb10-j0_1 with IP 10.0.1.10 has succesfully set: cycletime = 1000 usec, RX DO TX = ( 400 300 300 ) usec and TX rate = 5  every cycle
[INFO] EthResource::askBoardVersion() found BOARD right_leg-eb10-j0_1 @ IP 10.0.1.10 of type ems4 with FW = ver 3.37 built on 2021 Feb 19 15:30
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 85s 579m 365u: (code 0x05000015, par16 0x2000 par64 0x0000000000ff0000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
[INFO]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1) @ 85s 579m 559u: CAN discovery has started for 2 eobrd_foc boards on (can1map, can2map) = (0x0006, 0x0000) with target can protocol ver 1.6 and application ver 3.3.3.
[WARNING]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1) @ 85s 580m 139u: CAN discovery has detected a eobrd_foc board in CAN1 addr 1 with can protocol ver 1.6 and application ver 3.3.4 Search time was 0 ms
[WARNING]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1) @ 85s 580m 289u: CAN discovery has detected a eobrd_foc board in CAN1 addr 2 with can protocol ver 1.6 and application ver 3.3.4 Search time was 0 ms
[INFO]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1) @ 85s 580m 448u: CAN discovery is OK for 2 eobrd_foc boards with target can protocol ver 1.6 and application ver 3.3.3. Search time was 0 ms
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 85s 580m 807u: (code 0x0500000d, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 85s 634m 927u: (code 0x04000001, par16 0x000f par64 0x0000010000000040) -> DEBUG: tag01 + CURRENT PID
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 85s 635m 40u: (code 0x04000001, par16 0x000f par64 0x0000018000000200) -> DEBUG: tag01 + VELOCITY PID
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 1, time 85s 649m 640u: (code 0x04000001, par16 0x000f par64 0x0000010000000040) -> DEBUG: tag01 + CURRENT PID
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 1, time 85s 649m 751u: (code 0x04000001, par16 0x000f par64 0x0000018000000200) -> DEBUG: tag01 + VELOCITY PID
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 86s 194m 52u: (code 0x04000001, par16 0x0000 par64 0x0000000000000000) -> DEBUG: tag01 + CALIB 12 j 0: offset=13452 zero=-32768 
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src CAN1, adr 1, time 90s 193m 7u: (code 0x04000002, par16 0x8119 par64 0x0000000011bc0000) -> DEBUG: tag02 + .
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 90s 193m 494u: (code 0x04000001, par16 0x0001 par64 0x0000000000000001) -> DEBUG: tag01 + SET CONTROLMODE
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src CAN1, adr 1, time 91s 385m 992u: (code 0x04000002, par16 0x8119 par64 0x0000000000000000) -> DEBUG: tag02 + .
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src CAN1, adr 1, time 92s 891m 994u: (code 0x04000002, par16 0x8119 par64 0xffcc0042ffff0079) -> DEBUG: tag02 + .
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src CAN1, adr 1, time 93s 120m 993u: (code 0x04000002, par16 0x8119 par64 0xffcc004238450079) -> DEBUG: tag02 + .
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 1, time 93s 451m 51u: (code 0x04000001, par16 0x0000 par64 0x0000000000000000) -> DEBUG: tag01 + CALIB 12 j 1: offset=59443 zero=-32768 
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src CAN1, adr 2, time 100s 482m 15u: (code 0x04000002, par16 0x8129 par64 0x0000000033540000) -> DEBUG: tag02 + .
[DEBUG]  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 1, time 100s 482m 503u: (code 0x04000001, par16 0x0001 par64 0x0000000000000001) -> DEBUG: tag01 + SET CONTROLMODE
[... cut ...]
[DEBUG]  from BOARD 10.0.1.12 (right_leg-eb12-j4_5), src CAN1, adr 2, time 134s 746m 145u: (code 0x04000002, par16 0x8129 par64 0x0000000000000000) -> DEBUG: tag02 + .
[DEBUG]  from BOARD 10.0.1.12 (right_leg-eb12-j4_5), src CAN1, adr 2, time 134s 752m 146u: (code 0x04000002, par16 0x8129 par64 0x00000000ffff0000) -> DEBUG: tag02 + .
tumme@pop-os:~/Documents$ 
```

After the message `[DEBUG] yarprobotinterface running happily` no more messages of that kind are printed also waiting few minutes

Here the entire log
[log.txt](https://github.com/robotology/icub-tech-support/files/6210745/log.txt)

davidetome, 2021-03-26:
> I found:
> 
> * Left leg : LS0 (AMO joint 0) w/ connector disconnected
> * Right Leg : RS1 (AMO joint 1) the red wire (positive pwr) uncrimped
> 
> cc @simeonedussoni

Left Leg joint 0:
@simeonedussoni moved the magnetic ring of the AMO (there was a little damage in a part of it) and re-calibrated the sensor.
We also replaced the wiring from the EMS to the sensor.
I re-calibrated the joint and now the faulty reading of the sensor should be resolved.

Right Leg joint 1:
The right leg is now working properly after crimping the broken cable.

Still pending the calibration of the motors offset.

cc @pattacini

S-Dafarra, 2021-03-31:
Unfortunately, this is still happening. I turned on the robot, left it still for some minutes, and then the left hip pitch joint went into HF. Actually, when I looked back at the robot, the joint position was ok, but checking at the logger I saw the same errors regarding the joint being outside the boundaries.

pattacini, 2021-03-31:
Thanks for reporting @S-Dafarra 👍🏻 
Tomorrow morning, we'll have a short internal meeting to understand what is happening that is dodging our analyses.
If possible, we'll take the robot soon afterward.

S-Dafarra, 2021-03-31:
> Thanks for reporting @S-Dafarra 👍🏻
> Tomorrow morning, we'll have a short internal meeting to understand what is happening that is dodging our analyses.
> If possible, we'll take the robot soon afterward.

Thanks a lot @pattacini. I wonder if it is board resetting after a while.

S-Dafarra, 2021-03-31:
I have to say that this is indeed quite puzzling. That joint is now ok since half an hour. I have only moved it slightly, but before it happened twice in a shorter time horizon without even moving :thinking:

pattacini, 2021-03-31:
Thanks for the further info @S-Dafarra 👍🏻 

Can you confirm that the error causing the HF is still the following?

> joint being outside the boundaries.

S-Dafarra, 2021-03-31:
> Can you confirm that the error causing the HF is still the following (when it happens)?
> 
> > joint being outside the boundaries.

Yes. Here the corresponding log.

[log_icub-head_yarprobotinterface_1748_left_hip_pitch_again.txt](https://github.com/robotology/icub-tech-support/files/6238272/log_icub-head_yarprobotinterface_1748_left_hip_pitch_again.txt)

S-Dafarra, 2021-03-31:
I remember that at startup the right shoulder roll did not started correctly, but I was able to start it from the motorgui. I did not notice it if it occurred also afterwards.

simeonedussoni, 2021-04-01:
we ran oll de morning the robot, logging readings from the encoders
the roboto never went into HF apart from a trigger due to cable unplugging
I analyzed the logged data and found no problems. 
At a first glance it seemed to me that there were steps in measured angle as high as $\~50^{\circ}$ but at a deeper look it turned out to be due to some crazy matlab import tool behaviour: for instance, we can see that some lines from the log files
[image]

have been misinterpreted in the matlab imported variable:
[image]

giving origin to peaks in the derivative of the encoder readings.
thus we have not a decisive hint on what can happen 🤯 

we ask @S-Dafarra to take logs of the encoder diuring their next experiments in order to track the raw data if the failure is raised again.

[5 more replies]

closed 2021-05-01 by pattacini