robotology/mesh-support #1104: pc104 not always booting. RTC/cmos battery too old ?

EliseiFrederic, 2021-03-19:
## Brief description of the request/failure
pc104 of icubGrenoble01 tends to fail booting more and more often. If powered for some time (say more than 1 or 2 hours rather than just a few minutes), seems to succeed everytime. But then, 2 days later, getting first successful boot is not immediate.

## Detailed context
We don't yet have a VGA cable to connect to see if there is a BIOS error, waiting for a keypress.
The robot has been used less often with the lock-out. Maybe the RTC/CMOS battery is just too old (more than 8 years now)
Reported RTC time is wrong after a successfull boot :
[log_20210317_165300_proc_deriver_rtc.txt](https://github.com/robotology/icub-tech-support/files/6171620/log_20210317_165300_proc_deriver_rtc.txt)

boot time in last command show the same problem since a few boot...

Should I change this battery ? Is it a CR2032 like many PC motherboards ? Easy to access ?
Some experiments are running, so I could not open the head yet.
Or should we search another reason for this erratic fail-to-boot (and build a VGA cable)

Thanks in advance,
Frédéric

:~$ last | more
icub     pts/1        192.168.70.33    Fri Mar 19 15:14   still logged in
icub     pts/1        192.168.70.52    Fri Mar 19 14:27 - 14:28  (00:00)
icub     pts/0        192.168.70.52    Thu Nov  3 18:16    gone - no logout
reboot   system boot  4.9.0-4-rt-amd64 **Thu Jan  3** 00:05   still running
icub     pts/1        192.168.70.52    Fri Mar 19 09:38 - 09:38  (00:00)
icub     pts/0        192.168.70.52    Fri Mar 19 09:33 - 11:17  (01:44)
reboot   system boot  4.9.0-4-rt-amd64 **Wed Jan  2** 19:12 - 11:17 (7015+16:05)
icub     pts/1        192.168.70.52    Thu Mar 18 16:05 - 16:05  (00:00)
icub     pts/1        192.168.70.52    Thu Mar 18 14:14 - 14:14  (00:00)
icub     pts/0        192.168.70.52    Thu Mar 18 14:10 - 16:06  (01:56)
icub     pts/0        192.168.70.52    Thu Mar 18 10:40 - 13:06  (02:25)
reboot   system boot  4.9.0-4-rt-amd64 **Tue Jan  1** 20:19 - 16:06 (7015+19:46)
icub     pts/0        192.168.70.52    Wed Mar 17 17:12 - 17:23  (00:11)
icub     pts/0        192.168.70.52    Wed Mar 17 16:51 - 17:12  (00:20)
reboot   system boot  4.9.0-4-rt-amd64 **Tue Jan  1** 02:30 - 17:23 (7015+14:53)
icub     pts/0        192.168.70.52    Mon Mar 15 15:47 - 15:54  (00:07)
reboot   system boot  4.9.0-4-rt-amd64 **Tue Jan  1** 01:40 - 15:54 (7013+14:14)
icub     pts/0        192.168.70.52    Thu Nov  3 18:18 - 18:21  (00:02)
reboot   system boot  4.9.0-4-rt-amd64 **Tue Jan  1** 01:33 - 18:21 (5420+16:47)
icub     pts/1        192.168.70.52    Fri Mar  5 14:24 - 14:24  (00:00)
icub     pts/0        192.168.70.52    Fri Mar  5 15:45 - 15:02  (00:-43)
reboot   system boot  4.9.0-4-rt-amd64 Fri Mar  5 15:45 - 15:02  (00:-43)
icub     pts/1        192.168.70.52    Thu Mar  4 09:45 - 10:59  (01:14)
icub     pts/0        192.168.70.52    Thu Mar  4 11:08 - 11:01  (00:-6)
reboot   system boot  4.9.0-4-rt-amd64 Thu Mar  4 11:07 - 11:01  (00:-6)
icub     pts/1        192.168.70.52    Thu Mar  4 09:39 - 09:41  (00:02)
icub     pts/1        192.168.70.52    Thu Mar  4 09:38 - 09:38  (00:00)
icub     pts/0        192.168.70.52    Thu Mar  4 09:36 - 09:41  (00:05)
reboot   system boot  4.9.0-4-rt-amd64 Thu Mar  4 10:59 - 09:41  (-1:-17)
icub     pts/0        192.168.70.52    Wed Mar  3 09:27 - 10:57  (01:30)
reboot   system boot  4.9.0-4-rt-amd64 Wed Mar  3 10:50 - 10:57  (00:07)
icub     pts/0        192.168.70.52    Mon Mar  1 14:28 - 15:55  (01:26)
reboot   system boot  4.9.0-4-rt-amd64 Mon Mar  1 15:52 - 15:55  (00:02)
icub     pts/1        192.168.70.52    Mon Mar  1 09:27 - 09:27  (00:00)
icub     pts/0        192.168.70.52    Mon Mar  1 09:25 - 11:03  (01:38)
reboot   system boot  4.9.0-4-rt-amd64 Mon Mar  1 10:47 - 11:03  (00:16)

## Accompanying material

EliseiFrederic, 2021-03-19:
Looks like it's soldered in place :
[image]

pattacini, 2021-03-22:
Hi @EliseiFrederic 

This week, our staff is busy with the tests required to deliver a new robot, so they won't be available soon to give you support on that.

However, I managed to collect some initial impressions from what you wrote above and the recommendation is that you plug the VGA to see what's going on underneath.

You could also replace the old battery with a brand new one you have to solder anyway (this is the [datasheet of the ATOM][1] board), although it's unlikely that this is causing the problem since you would have seen each cold boot failing.

cc @mbrunettini  

[1]: https://github.com/robotology/icub-tech-support/files/6180600/EMBEDDED.LOGIC.-.PB-ATOM-L__N455_D525_Rev1.1.pdf

EliseiFrederic, 2021-03-22:
Thanks Ugo ! We made a 10 wires VGA cable, but looking at the datasheet that you posted, it looks like we need a 16 wires one for the ATOM board... 

You could also replace the old battery with a brand new one you have to solder anyway (this is the [ https://github.com/robotology/icub-tech-support/files/6180600/EMBEDDED.LOGIC.-.PB-ATOM-L__N455_D525_Rev1.1.pdf | datasheet of the ATOM ] board), although it's unlikely that this is causing the problem since you would have seen each cold boot failing. 

cc [ https://github.com/mbrunettini | @mbrunettini ]

EliseiFrederic, 2021-03-22:
We got a working 16 wires cable to connect through VGA.
No error message on BIOS yet (but the power supply was left "on" between experiments since a few days).
We'll revert to the normal power supply "off" mode, and try to power on the robot every day until we see the lock condition/error message appearing.

EliseiFrederic, 2021-03-23:
Surprisingly, on cold boots, the AMIBIOS fails to boot on USB or tries another device...
**"Reboot and Select proper Boot device
or Insert Boot Media in selected Boot Device and press a key"**
We have to find what is the related boot device... If it's still USB, maybe a BIOS option (delay before booting on USB) has been lost with the low-level battery ?

EliseiFrederic, 2021-03-23:
Previous message includes this:
**"CMOS Settings Wrong
CMOS Date/Time Not Set
Press F1 to Run SETUP
Press F2 to load default values"**
and the clock was reverted to 2002-01-01 (as seen a few hours later when keeping CPU PSU on to allow normal boot)

It seems that keeping PSU "on" makes the clock run again, so the "CMOS settings wrong" flag is not set anymore on the next warm boot, and the boot sequence on the USB key runs as expected...

EliseiFrederic, 2021-03-23:
[image]

pattacini, 2021-03-25:
From your analyses, it seems that replacing the battery could improve the situation.
The battery needs to be soldered though.

reafrancesco, 2021-03-25:
Dear Friends,
we recently had a similar issue (the pc104 was not starting or when removed can bus connections was not able to complete the boot) with the PC104 but I don`t know if what is causing this it the same problem.
We thought that the issue might have been the battery and we changed but this did not fix the problem.
After weeks of testing with @julijenv we realized that it was an issue with the power supply of the PC104. We changed it and the PC104 started to work as before. I don`t know if this is the same for you but I would rather check the PS if I were you.

EliseiFrederic, 2021-03-25:
Thank you @pattacini . We'll try that as soon as we have a slot between 2 experiments.
We will have to remove/unconnect parts of the motherboard. Any advice to avoid errors/break things

EliseiFrederic, 2021-03-25:
@reafrancesco Thanks for your advice. We have this strange behavior where having the PSU "on" between the sessions where we use the robot is enough to have it to boot. Not doing so, it looses the date and wait at BIOS boot time.
Did you have such symptoms too ?
Thanks again for your message

reafrancesco, 2021-03-25:
Actually we have never lost the date. Perhaps you are right this makes us think that this is more related to the battery. Anyway what I wanted to tell you is that we removed the battery and changed with a new one so we have experience in the process. We can also give you some insights in the successful process for us.
Please do not hesitate to contact me if you need details.

EliseiFrederic, 2021-06-07:
Tech guys here soldered a socket, accessible from the front side.
On the back of the PC104 board, where the soldered battery was, they had to add a connecting wire, as the battery was connecting to two points at the same time (two +5V I think), over a chip (was hidden under the original battery).
Date and power-on seems ok since ! So, closing the issue, thanks.

closed 2021-06-07 by EliseiFrederic

pattacini, 2021-06-07:
Awesome @EliseiFrederic!
Happy to know that you managed to get it to work again.