robotology/mesh-support #1105: Right knee quadrature encoder broken

S-Dafarra, 2021-03-22:
## Brief description of the request/failure
I just tried to run the robot, but the right knee does not move well. It can move for a bit, but then it gets stuck.

There are also other problems on the left hip pitch and roll, for which #1103 is already opened.

## Detailed context

## Accompanying material

[log_icub-head_yarprobotinterface_2065_left_hip_right_knee_broken.txt](https://github.com/robotology/icub-tech-support/files/6181399/log_icub-head_yarprobotinterface_2065_left_hip_right_knee_broken.txt)

S-Dafarra, 2021-03-22:
cc @simeonedussoni @davidetome @emilianob80 @pattacini

S-Dafarra, 2021-03-22:
After moving it a bit back and forth, it started working again. It seems a bit weird.

pattacini, 2021-03-22:
@simeonedussoni will be checking #1103, while for this we need to understand how to proceed as other members are busy w/ the new robot or are not at CRIS.

>After moving it a bit back and forth, it started working again. It seems a bit weird.

Definitely weird, yep. Needing investigation.

violadelbono, 2021-03-22:
@S-Dafarra are the values of the knee joint from yarpmotorgui consistent? If yes, I think it could be a loose/partially broken cable

S-Dafarra, 2021-03-22:
> @S-Dafarra are the values of the knee joint from yarpmotorgui consistent? If yes, I think it could be a loose/partially broken cable

The joint encoder seemed ok, but I forgot to check the motor encoder.

S-Dafarra, 2021-03-23:
> @S-Dafarra are the values of the knee joint from yarpmotorgui consistent? If yes, I think it could be a loose/partially broken cable

With @davidetome and @simeonedussoni we also quickly checked the motor encoder, but it seemed to be fine. Often, after it gets stuck, it is sufficient to move the knee to have it back online. I think the connection might be a bit loose.

pattacini, 2021-03-24:
From https://github.com/robotology/icub-tech-support/issues/1103#issuecomment-804907101, it seems that this has been somehow solved. Is this understanding correct?

pattacini, 2021-03-24:
~~From #1103 (comment), it seems that this has been somehow solved. Is this understanding correct?~~

Nope ❌ as per https://github.com/robotology/icub-tech-support/issues/1103#issuecomment-805648152.

pattacini, 2021-03-30:
This has been solved, hence closing.
Open it again if needed.

closed 2021-03-30 by pattacini