robotology/mesh-support #1106: Error opening yarpdev in 'Face' app

roalchaq, 2021-03-22:
Hi there!,

When running the Face application on `iCubLondon01` I get the following error in particular with the `yarpdev` module:

```console
$ yarpdev --name /icub/face/raw --device serial --subdevice serialport --context faceExpressions --from serialport.ini --verbose

||| configuring
||| added context faceExpressions
||| default config file specified as serialport.ini
||| checking [/home/icub/serialport.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubLondon01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubLondon01] (robot YARP_DATA_HOME)
||| found /home/icub/.local/share/yarp/robots/iCubLondon01
||| checking [/etc/yarp/robots/iCubLondon01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/install/yarp/share/yarp/robots/iCubLondon01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/install/icub-main/share/iCub/robots/iCubLondon01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/install/icub-contrib-common/share/ICUBcontrib/robots/iCubLondon01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/install/icub-contrib-common/share/ICUBcontrib/robots/iCubLondon01
||| checking [/usr/local/src/robot/install/yarp/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/install/icub-main/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/install/icub-contrib-common/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/.local/share/yarp/robots/iCubLondon01/serialport.ini] (robot)
||| checking [/usr/local/src/robot/install/icub-contrib-common/share/ICUBcontrib/robots/iCubLondon01/serialport.ini] (robot)
||| checking [/home/icub/.config/yarp/contexts/faceExpressions] (context YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/contexts/faceExpressions] (context YARP_DATA_HOME)
||| checking [/etc/yarp/contexts/faceExpressions] (context YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/install/yarp/share/yarp/contexts/faceExpressions] (context YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/install/icub-main/share/iCub/contexts/faceExpressions] (context YARP_DATA_DIRS)
||| found /usr/local/src/robot/install/icub-main/share/iCub/contexts/faceExpressions
||| checking [/usr/local/src/robot/install/icub-contrib-common/share/ICUBcontrib/contexts/faceExpressions] (context YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/install/icub-main/share/iCub/contexts/faceExpressions/serialport.ini] (context)
||| found /usr/local/src/robot/install/icub-main/share/iCub/contexts/faceExpressions/serialport.ini
[DEBUG]Subdevice serialport
yarp: Port /log/pc104/yarpdev/3111 active at tcp://10.0.2.2:10075/
yarp: Sending output from /log/pc104/yarpdev/3111 to /yarplogger using fast_tcp
[INFO]Starting Serial Port in /dev/ttyACM0 
[ERROR]Can not control communications port /dev/ttyACM0 
yarpdev: ***ERROR*** driver <serialport> was found but could not open
[ERROR]cannot make <serialport>
yarpdev: ***ERROR*** driver <serial> was found but could not open
===============================================================
== Options checked by device:
== 
device=serial
==
===============================================================
[ERROR]yarpdev: ***ERROR*** device not available.
[INFO]Suggestions:
[INFO]+ Do "yarpdev --list" to see list of supported devices.
yarp: Removing output from /log/pc104/yarpdev/3111 to /yarplogger
```
`ttyACM0` was not being initially listed under `/dev`. However, after checking the input voltage to the board that handles the leds and making sure that the conection to the USB port was ok I can see the serial port being listed.

The documentation provided [here](https://icub-tech-iit.github.io/documentation/icub_starter_kits/first_steps/#robot-calibration) gives me the impression that the leds should light up just after launching `yarprobotinterface` in `pc104`, which does not happens with our robot. 

Any suggestions? Is there a way of testing that the board handling the leds is not broken?

pattacini, 2021-03-25:
Hi @roalchaq 

Is `yarpdev` starting correctly now?
If so, you may be testing it by launching the script [`cycle.sh`][1].

[1]: https://github.com/robotology/icub-main/blob/master/app/faceExpressions/scripts/cycle.sh

roalchaq, 2021-03-25:
Hi @pattacini,

Unfortunately not. `yarpdev` is not starting. I get the error described in my first comment:

```console
[ERROR]Can not control communications port /dev/ttyACM0 
yarpdev: ***ERROR*** driver <serialport> was found but could not open
[ERROR]cannot make <serialport>
yarpdev: ***ERROR*** driver <serial> was found but could not open
===============================================================
== Options checked by device:
== 
device=serial
==
===============================================================
[ERROR]yarpdev: ***ERROR*** device not available.
```

davidetome, 2021-03-31:
Ciao @roalchaq ,

 here it is a simple procedure to test the facial expressions using a PC.

- Connect the board to a Windows PC's USB
-  When asked to, install the driver : [inf.zip](https://github.com/robotology/icub-tech-support/files/6235681/inf.zip)
- Check which `COM` is assigned by windows
-  Launch a serial terminal (i.e. https://www.hw-group.com/software/hercules-setup-utility) 
- In the serial terminal set : 
  1. COM port
  2. Buad rate: 9600 
  3. Data size: 8
  4. Parity bit: none
- Click on Open
- Send the command `I` and you should see all the leds on and the servo motor moving

Let us know if this test works properly

cc @pattacini

roalchaq, 2021-03-31:
Hi @davidetome,

Thank you for your help with this. I followed your procedure and even though I managed to open the serial port and send the `I` command, nothing happened with the leds and the servo did not move. I guess this means the board is definitely broken.

davidetome, 2021-03-31:
> Hi @davidetome,
> 
> Thank you for your help with this. I followed your procedure and even though I managed to open the serial port and send the `I` command, nothing happened with the leds and the servo did not move. I tried the procedure with pc104 on and off. I guess this means the board is definitely broken.

The procedure described above is intended to be used w/ the face/face -expressions on the desk and detached from the `pc104`.

Could you please detach the face from the `pc104`, connect the USB coming from the face to the PC and check w/ a multimeter if on pins 1-5 of the PROG connector on the face expressions board you measure 5Vdc?

roalchaq, 2021-03-31:
The usb port was detached from `pc104`  and connected to a windows computer. Just the 12 V power cables were still connected to `pc104`, thus in case the power coming from the computer's usb port wasn't enough. I am sorry if I didn't explain my self correctly. I made a second test with the face completely detached from the robot (USB and 12 V power input) but it didn't work either. I got 5 V between pins 1 and 5 of the prog connector.

davidetome, 2021-04-01:
>  am sorry if I didn't explain my self correctly

Perfectly clear! Actually, the 12V is needed only for the servo motor, without it you should be able to send commands to the board and turn on the LEDs.

Anyway, Let me check on our setup (if available) and I'll give you feedback asap.

davidetome, 2021-04-07:
Ciao @roalchaq ,

 I finally successfully tested the above procedure, so my opinion is that your face expression control board Is broken. 

https://user-images.githubusercontent.com/6638215/113851120-73980600-979b-11eb-8f1a-538f5506f6d0.mp4

At this point, you may need to replace it, let us know.
Keep in mind that w/ a new board you may need to re-calibrate the eyelids opening range (servo motor) as described below:

- Send (via serial terminal) the command `EXX` (where XX -> 0-99 and it sets the `max` position accordingly the mechanical constraints of your face cover)
- Send (via serial terminal) the command `UXX `(where XX -> 0-99 and it sets the `min `position accordingly the mechanical constraints of your face cover)

I'm preventively asking my peer @Fabrizio69 , if we have one available to eventually ship (`IIT code 736`).

cc @pattacini

roalchaq, 2021-04-07:
Hi @davidetome ,

Thank you again for your help with this. We are defenitely interested in replacing the board. Please let me know how to proceed.

davidetome, 2021-04-08:
> Hi @davidetome ,
> 
> Thank you again for your help with this. We are defenitely interested in replacing the board. Please let me know how to proceed.

👍🏻  So you can refer to @Fabrizio69 who manage the production/warehouse.

pattacini, 2021-04-09:
Rather than @Fabrizio69 , it'd be more convenient to get in touch w/ [Andrea Derito][1], who's responsible for the contacts w/ our customers.

cc @andreaderito

[1]: https://www.iit.it/web/guest/people-details/-/people/andrea-derito

andreaderito, 2021-04-09:
Thank you @pattacini and @davidetome

andreaderito, 2021-04-09:
Dear @roalchaq 
please be so kind as to provide me via e-mail (andrea.derito@iit.it) with the following information:
Name of institution to whom I may address the quotation
Department (if applicable)
Full address
Ref. person (the quotation will be addressed to that person and to you, of course)
e-mail of that person
Furthermore kindly propvide me with:
1) delivery address;
2) Ref. person to receive the goods
3) phone number and email address of the person receiving the goods.
As soon as I get this e-mail I may prepare a formal quotation for you.
Thx in advance, best
PS: I ask for e-mail because when sending out the quotation I should also include the history of the request for quotation so the responsible/supervisor/budget owner from your institution is able to understand/rebuild the origin of the quotation. Thank you for cooperating.

pattacini, 2021-04-10:
Closing this support ticket then.

andreaderito, 2021-04-10:
Hi @roalchaq thank you for your email with the request for quotation, the good piece of news is that we have availability of the spare and, since the value is limited, our boss, @maggia80 decided to send it for free with no need for quotation, purchase order, acceptance, invoicing, payment and so on. I will see if I may also ship it as free sample due to the low value so you avoid paying the customs clearance fees/taxes... I will follow this up by email, best

andreaderito, 2021-06-21:
Customer confirmed reception of the spare

cedricgoubard, 2021-07-12:
Hi @davidetome,
I work with @roalchaq on our iCub, and unfortunately the board was apparently not the issue. We managed to get the face working with both boards (the old and the new one) by plugging the serial chord into another laptop and using the serial.sh script you provided.

In other words, the issue seems to be coming from pc104. The error message is always the same:

davidetome, 2021-07-15:
Ciao @cedricgoubard ,

 is the `/dev/ttyACM0` device created when you plug the board in the head's USB port?
Check it simply doing `ls -la /dev/tty*` .

cedricgoubard, 2021-07-15:
Hi @davidetome,
Yes, the device is detected (output of `ls -la /dev/tty*`):

[image]

However, yarpdev is still having the same issue:

[image]

davidetome, 2021-07-16:
Hi @cedricgoubard , 

 could you check if in `/etc/groups` file `icub` is in the `tty` group?

[image]

cc @pattacini

cedricgoubard, 2021-07-16:
Hi @davidetome,
Thank you for following this up with me. Apparently iCub was not in the group:
[image]

I added it and restarted pc104:
[image]

But I am still getting the same error:
[image]

I even tried to change the permissions, but the message remains the same:
[image]

cedricgoubard, 2021-07-16:
Hi @davidetome,
We plugged the face to the pc104 (both USB and 12V), and we tried to use the serial port:
`echo -e 'I' > /dev/ttyACM0`
That gave us the following result: https://photos.app.goo.gl/Q2BWDwHAfqjx443L7

We also tried to command the face on a laptop, and the exact same yarpdev command with the same context worked. We encountered another issue when starting emotionsInterface, but I think this may be related to the yarpdev version we have on the laptop.

So I guess the issue is probably software-related, since the hardware seems to work? We have an old iCub (3rd ever made) with a 32bits PC104, if that helps.

pattacini, 2021-07-19:
Thank you @cedricgoubard for the enlightening test! 💡

There must be then some problems in the middleware.
We will try to get back to you asap on this.

In the meantime, could you report to us the output of the following commands launched on the `pc104` hub?
```console
lsb_release -a
yarp --version
cd $YARP_ROOT && git log -1
yarpdev --list
```

Also, it'd be good to know the content of the file you can retrieve by launching the following command on the `pc104`:
```
yarp resource --context faceExpressions --from serialport.ini
```

cedricgoubard, 2021-07-20:
Hi @pattacini,
Here is the information you asked for. I took the liberty of including the last 2 commits for the `$YARP_ROOT` dir, since the last one seems to be only present locally.
```bash
icub@pc104:~$ lsb_release -a
LSB Version:	core-2.0-ia32:core-2.0-noarch:core-3.0-ia32:core-3.0-noarch:core-3.1-ia32:core-3.1-noarch:core-3.2-ia32:core-3.2-noarch:core-4.0-ia32:core-4.0-noarch:core-4.1-ia32:core-4.1-noarch:security-4.0-ia32:security-4.0-noarch:security-4.1-ia32:security-4.1-noarch
Distributor ID:	iCub Debian
Description:	Debian GNU/Linux 7.2 (wheezy) + iCub customizations
Release:	7.2
Codename:	wheezy
```
```bash
icub@pc104:~$ yarp version
YARP version 3.1.0+192-20190206.2+git81ca0b0
```
```bash
icub@pc104:~$ cd $YARP_ROOT && git log -n 2
commit 5734917c6ede6387c59171c3cfcc048e303d9ffb
Author: Tobias Fischer <t.fischer@imperial.ac.uk>
Date:   Tue May 28 19:03:48 2019 +0200

    32-bit compatibility changes

commit 81ca0b0458d66bdb1f48edf3f1267a136cfe8ca1
Author: Daniele E. Domenichelli <daniele.domenichelli@iit.it>
Date:   Wed Feb 6 14:11:54 2019 +0100

    Examples: Fix -Wformat warning
```

<details>
<summary>yardev --list output</summary>

```bash
icub@pc104:/usr/local/src/robot/yarp2$ yarpdev --list
[INFO]Here are devices listed for your system:
yarp: Cannot connect to port /root
yarp: Port /log/pc104/yarpdev/2957 failed to activate (invalid address)
LogForwarder error while opening port /log/pc104/yarpdev/2957
yarp: No connection to nameserver
yarp: *** try running: yarp detect ***
[INFO]Device "test_motor", C++ class yarp::dev::TestMotor, wrapped by "controlboard"
[INFO]Device "serial", C++ class yarp::dev::ServerSerial, is a network wrapper.
[INFO]Device "remote_grabber", C++ class yarp::dev::RemoteFrameGrabber, wrapped by "grabber"
[INFO]Device "grabber", C++ class yarp::dev::ServerFrameGrabber, is a network wrapper.
[INFO]Device "pipe", C++ class yarp::dev::DevicePipe, has no network wrapper
[INFO]Device "group", C++ class yarp::dev::DeviceGroup, has no network wrapper
[INFO]Device "remote_controlboard", C++ class yarp::dev::RemoteControlBoard, wrapped by "controlboardwrapper2"
[INFO]Device "inertial", C++ class yarp::dev::ServerInertial, is a network wrapper.
[INFO]Device "controlboardwrapper2", C++ class yarp::dev::ControlBoardWrapper2, is a network wrapper.
[INFO]Device "analogsensorclient", C++ class AnalogSensorClient, has no network wrapper
[INFO]Device "analogServer", C++ class yarp::dev::AnalogWrapper, is a network wrapper.
[INFO]Device "virtualAnalogServer", C++ class yarp::dev::VirtualAnalogWrapper, is a network wrapper.
[INFO]Device "RGBDSensorWrapper", C++ class yarp::dev::RGBDSensorWrapper, is a network wrapper.
[INFO]Device "RGBDSensorClient", C++ class yarp::dev::RGBDSensorClient, is a network wrapper.
[INFO]Device "controlboardremapper", C++ class yarp::dev::ControlBoardRemapper, wrapped by "controlboardwrapper2"
[INFO]Device "remotecontrolboardremapper", C++ class yarp::dev::RemoteControlBoardRemapper, wrapped by "controlboardwrapper2"
[INFO]Device "robotDescriptionServer", C++ class RobotDescriptionServer, has no network wrapper
[INFO]Device "robotDescriptionClient", C++ class RobotDescriptionClient, has no network wrapper
[INFO]Device "grabberDual", C++ class yarp::dev::ServerGrabber, is a network wrapper.
[INFO]Device "JoypadControlServer", C++ class yarp::dev::JoypadControlServer, is a network wrapper.
[INFO]Device "JoypadControlClient", C++ class yarp::dev::JoypadControlClient, is a network wrapper.
[INFO]Device "Rangefinder2DClient", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_Rangefinder2DClient.so library).
[INFO]Device "Rangefinder2DWrapper", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_Rangefinder2DWrapper.so library).
[INFO]Device "ServerSoundGrabber", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_ServerSoundGrabber.so library).
[INFO]Device "batteryClient", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_batteryClient.so library).
[INFO]Device "batteryWrapper", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_batteryWrapper.so library).
[INFO]Device "localization2DClient", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_localization2DClient.so library).
[INFO]Device "map2DClient", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_map2DClient.so library).
[INFO]Device "map2DServer", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_map2DServer.so library).
[INFO]Device "multipleanalogsensorsclient", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_multipleanalogsensorsclient.so library).
[... 91 lines cut ...]
```
/// A initialization file for configuring chico facial expressions interface
comport /dev/ttyACM0
baudrate 9600
xonlim 0
xofflim 0
readmincharacters 1
readtimeoutmsec 0
parityenb 0
paritymode EVEN
ctsenb 0
rtsenb 0
xinenb 0
xoutenb 0
modem 0
rcvenb 0
dsrenb 0
dtrdisable 0
databits 7
stopbits 1
```

</details>

Let me know if you need anything else, and thanks again for your help!

pattacini, 2021-07-20:
Thanks heaps @cedricgoubard 

Something a bit strange that comes to my eyes is the following:

```bash
[INFO]Device "serialport", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_serial.so library), wrapped by "serial".
[INFO]Device "serialport", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_serialport.so library), wrapped by "serial".
```

It seems that you've got two libraries for this service. Maybe the oldest version is shadowing the correct one causing the problem.

The correct one should be that pointing to `yarp_serial.so`, so you may delete `yarp_serialport.so` and try again.
(you may also think of comparing the dates of the two libraries and/or clean them up and issue a new YARP build).

Well, take backup copies of those two libraries anyway.

cedricgoubard, 2021-07-20:
Thanks @pattacini! 
I think you are onto something: I removed the `yarp_serialport.so`, and now something new happens.
When I start pc 104, I can still get the face to react by using my `echo -n 'I' > /dev/ttyACM0`.
Then, when I run yarpdev, the error is different from the one we had previously: 

```bash
icub@pc104:~$ yarpdev --name  /icub/face/raw --device serial --subdevice serialport --context faceExpressions --from serialport.ini
[DEBUG]Subdevice serialport
yarp: Port /log/pc104/yarpdev/4002 active at tcp://10.0.2.2:10247/
yarp: Sending output from /log/pc104/yarpdev/4002 to /yarplogger using fast_tcp
[INFO]Starting Serial Port in /dev/ttyACM0 
[ERROR]Invalid communications port in /dev/ttyACM0: Input/output error
yarpdev: ***ERROR*** driver <serialport> was found but could not open
[ERROR]cannot make <serialport>
yarpdev: ***ERROR*** driver <serial> was found but could not open
[ERROR]yarpdev: ***ERROR*** device not available.
[INFO]Suggestions:
[INFO]+ Do "yarpdev --list" to see list of supported devices.
[INFO]+ Or append "--verbose" option to get more information.
yarp: Removing output from /log/pc104/yarpdev/4002 to /yarplogger
```
Our `Cannot control communications on port /dev/ttyACM0` has now become `Invalid communications port in /dev/ttyACM0: Input/output error`.

The strange part is that after I try to run yarpdev, the `echo -n 'I' > /dev/ttyACM0` does not work anymore:
```bash
icub@pc104:~$ echo -n 'I' > /dev/ttyACM0 
-bash: /dev/ttyACM0: Input/output error
```

I think this might have something to do with a new warning that appears now when I run `yarpdev --list`:
```bash
icub@pc104:~$ yarpdev  --list | grep serial
yarp: Port /log/pc104/yarpdev/4057 active at tcp://10.0.2.2:10251/
yarp: Sending output from /log/pc104/yarpdev/4057 to /yarplogger using fast_tcp
[WARNING]Wrong library name for plugin serialport
[INFO]Device "serial", C++ class yarp::dev::ServerSerial, is a network wrapper.
[INFO]Device "serialport", available on request (found in /usr/local/src/robot/install/yarp/lib/yarp/yarp_serial.so library), wrapped by "serial".
yarp: Removing output from /log/pc104/yarpdev/4057 to /yarplogger
```

I checked, and port `ttyACM0` has not changed in `/dev`.
```bash
icub@pc104:~$ ls -l /dev/ttyACM0 
crw-rw---T 1 root dialout 166, 0 Jul 20 14:54 /dev/ttyACM0
```

I tried to see if a process was already using the port, but nothing comes up:
```bash
icub@pc104:~$ ls -l /proc/[0-9]*/fd/* |grep /dev/ttyACM0
ls: cannot access /proc/4049/fd/255: No such file or directory
ls: cannot access /proc/4049/fd/3: No such file or directory
ls: cannot access /proc/4050/fd/255: No such file or directory
```

Do you have any thoughts on this?

pattacini, 2021-07-20:
>[WARNING]Wrong library name for plugin serialport

This is because some YARP .ini files might still contain a reference to the old library you removed: see https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/yarp/dev/Drivers.cpp#L99.

Those files are located probably in `/usr/local/src/robot/install/share/yarp/plugins` (anyway, they're pointed at by `YARP_DATA_DIRS` env variable).

The "_Invalid communications port in /dev/ttyACM0: Input/output error_" si issued by:
https://github.com/robotology/yarp/blob/master/src/devices/serialport/SerialDeviceDriver.cpp#L49.

Honestly, given what we've spotted into `/usr/local/src/robot/install`, I would recommend that you:
- take a backup of `/usr/local/src/robot/install` – just in case;
- purge the content of `/usr/local/src/robot/install`;
- `make install` YARP again (and ICUB too);
- check the outcome.

cedricgoubard, 2021-07-21:
Hi @pattacini,
Thank you for the advice.
I am trying to reinstall everything, but I am facing a few issues.

I am using our [own fork of the superbuild (on the cpu-32 branch)](https://github.com/ImperialCollegeLondon/robotology-superbuild/tree/cpu-32), which was created by former members of the lab. From what I understand, it is basically of former version of your public repo + 1 commit to fix the tags of several dependencies, to match the requirements of our 32 bits pc104.

Here is my understanding of the issue, but let me know if I am wrong.
Since our pc104 is 32bits, we are stuck with Debian iCub wheezy (4), which only support gcc up to v4.7.
This means that we cannot use c++11, which prevents us from building many robotology repositories.

I managed to get YCM installed using ycm-0.10.2-offline.tar.gz (which is the oldest version that does not encounter [this issue](https://github.com/robotology/robotology-superbuild/issues/312)). 
Then, I tried building the superbuild, but yarp requires RTF 1.4.61; however, the latest version of RTF which does not require c++11 is 1.2.0.

Is there any way to know the exact version of each package which is compatible with our architecture?

traversaro, 2021-07-21:
Hi @cedricgoubard ! 

> Is there any way to know the exact version of each package which is compatible with our architecture?

Unfortunately we only started releasing coordinates tags of our software in February 2020, see https://icub-tech-iit.github.io/documentation/sw_versioning_table/ , so for getting the coordinated release before hand we need to do a bit of guess. 

However, in this specific case RTF is not a strictly required dependency of YARP or ICUB for most of their functionalities, so I think the easiest option is just to disable RTF, I tried to do it in https://github.com/ImperialCollegeLondon/robotology-superbuild/pull/1 , but I did not tested it, feel free to try and report if it still requires RTF even after this modifications.

traversaro, 2021-07-21:
> Is there any way to know the exact version of each package which is compatible with our architecture?

Just to double check, I checked back in the superbuild history and I switched the CI to stop using gcc 4.* in https://github.com/robotology/robotology-superbuild/pull/94 in June 2018, so any commit after that data could have problem on gcc 4.*, sorry about that. As you may imagine the only long term solution to this problem is https://github.com/robotology/icub-tech-support/issues/802, but feel free to let us know if there is anything we can do to help you even without the 64-bit CPU.

pattacini, 2021-07-22:
Hi @cedricgoubard 

If you feel adventurous, you could try out https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/the-icub-live-details.

These instructions are for building a live USB for the PC104, including the special CAN driver CFW002.
Notes are for the x64; however, it might be doable to adapt the guide for the 32-bit case.

Of course, if you'll do that, make the attempt with a brand new USB key while keeping the present one safe by your side.

cedricgoubard, 2021-07-23:
Hi @pattacini, @traversaro,

I removed RTF from the superbuild following @traversaro's advice, but I still encountered c++11 errors in YARP. I think this is because the YARP version we use in our superbuild is still too recent. 
Before going further back in our YARP version, I thought I'd try to compile g++ 4.8 from source to see if it fixes the issue. If it fails, I'll revert YARP to a commit before https://github.com/robotology/robotology-superbuild/pull/94, as @traversaro mentioned.

In the meantime, while g++ 4.8 is compiling, I think I'll try @pattacini's solution.
Thanks again for the help, I'll let you now how it goes!

cedricgoubard, 2021-07-27:
Hi,

Following @pattacini's advice, we managed to build a 32bits version of iCub live (Debian Buster)! pc104 seems to accept it, I am configuring it now.

In the meantime, I am facing another (small) issue: the board appears to boot only if I connect a keyboard and press 'enter'. Otherwise, it gets stuck on the grub (which is probably not really a grub since I did not find a /etc/default/grub file).

Do you have any idea on how to solve that?

pattacini, 2021-07-28:
>Following @pattacini's advice, we managed to build a 32bits version of iCub live (Debian Buster)! pc104 seems to accept it, I am configuring it now.

That's cool 👍🏻 

>In the meantime, I am facing another (small) issue: the board appears to boot only if I connect a keyboard and press 'enter'. Otherwise, it gets stuck on the grub (which is probably not really a grub since I did not find a /etc/default/grub file).

Just had a quick internal chat w/ @mbrunettini about that and it might be due to some settings in the BIOS, although it's strange that something happened in the BIOS by just changing the USB keys. Did you already have a look at it?

A somewhat related issue on the BIOS battery: #1104.

cedricgoubard, 2021-07-28:
I just found the solution: the `/usr/lib/live/mount/persistence/sda1/isolinux/isolinux.cfg` had a `timeout 0` option. I simply changed it to `timeout 5`, and it works.

I will now start to build YARP, iCub, etc. Should I use the YARP & iCub directories already present in `/usr/local/src/robot`, or clone the superbuild repo and start from there?

traversaro, 2021-07-28:
> I will now start to build YARP, iCub, etc. Should I use the YARP & iCub directories already present in `/usr/local/src/robot`, or clone the superbuild repo and start from there?

I would go for the robotology-superbuild path, especially if you are familar with that, and if you have enough disk space. : )

You can remove all the other repositories that are already in `/usr/local/src/robot`. For what regards the robotology-superbuild, remember to enable the `ROBOTOLOGY_ENABLE_ICUB_HEAD` and  `ROBOTOLOGY_USES_CFW2CAN` options, and disable the `ROBOTOLOGY_USES_GAZEBO` (see https://icub-tech-iit.github.io/documentation/sw_installation/icub_head_superbuild/ for details, at the moment we have a single page for both `icub-head` and `pc104`, but we plan to split this as tracked in https://github.com/icub-tech-iit/documentation/issues/149 .

traversaro, 2021-07-28:
By the way @cedricgoubard, if you have some modifications to the script to generate the icub-live to generate a 32-bit version it would be great if you could push them somewhere. I don't think we are going to support it officially and integrate it in our repo, but if other people are interested in that it could be useful.

pattacini, 2021-07-28:
In this respect, you may consider contributing to our [documentation](https://github.com/icub-tech-iit/documentation), @cedricgoubard.

cedricgoubard, 2021-07-28:
So, I also reinstalled the latest superbuild on our laptop, and I am now trying to launch icub and try a few features (ikart teleoperation, facial expressions, cameras, moving arms...) to make sure everything work.

However, I get the following error when launching `yarprobotinterface`:
```console
icub@pc104:~$ yarprobotinterface 
[DEBUG] Reading file /usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml
[WARNING] Invalid syntax while loading /usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml at line 2 . Unknown document type. Unknown url http://www.yarp.it/DTD/robotInterfaceV3.0.dtd
[WARNING] Invalid syntax while loading /usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml at line 2 . Unknown document type. Unknown url http://www.yarp.it/DTD/robotInterfaceV3.0.dtd
[WARNING] Invalid syntax while loading /usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml at line 2 . Unknown document type. Missing version in Url http://www.yarp.it/DTD/robotInterfaceV3.0.dtd
[WARNING] Invalid syntax while loading /usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml at line 2 . Unknown document type. Missing version in Url http://www.yarp.it/DTD/robotInterfaceV3.0.dtd
[WARNING] Invalid syntax while loading /usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml at line 1 . No DTD found. Assuming version yarprobotinterfaceV3.0
[ERROR] DTD V1.x has been deprecated. Please update your configuration files to DTD v3.x
[ERROR] Invalid DTD version, execution stopped.
[FATAL] Config file  /usr/local/src/robot/robotology-superbuild/build/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml  not parsed correctly.
Trace requested at /usr/local/src/robot/robotology-superbuild/src/YARP/src/yarprobotinterface/Module.cpp:136 by code called from:
/usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3(+0x715e5) [0xb7cc25e5]
/usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3(_Z16yarp_print_traceP8_IO_FILEPKcj+0x37) [0xb7cc5ed7]
yarprobotinterface(+0x5936) [0x40d936]
yarprobotinterface(+0x6ada) [0x40eada]
/usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0x61) [0xb7d11e51]
yarprobotinterface(+0x54aa) [0x40d4aa]
/lib/i386-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0xb77c4b41]
yarprobotinterface(+0x5531) [0x40d531]
```

Any idea what could cause that?

pattacini, 2021-07-28:
Hi @cedricgoubard 

Judging from your report, I believe there still were references to `robotinterface`, whereas we migrated to the use of `yarprobotinterface`.

I've applied a couple fo fixes:
- https://github.com/robotology/robots-configuration/commit/b93ab222e77950940a5044509f9355faa239e1fc
- https://github.com/robotology/robots-configuration/commit/a16a552b20a009a184531b66c042216bcabf89a3

Don't know if they're sufficient, but will certainly help out.

Could you please make an update and try again?

cedricgoubard, 2021-07-28:
Hi @pattacini, @traversaro,

It worked! I can now use facial expressions: https://photos.app.goo.gl/ERSvRjGUEtmb5ef26
I also checked ikart commands with the joystick and skin sensors, and both work.

**Thank you so much for your help!**

I'll leave the issue open until I've documented the process to create a 32bits Debian image, but feel free to close it if you want.
I also have several other hardware issues to fix in the weeks to come (arm sensors not working, no speaker/microphones...), but I'll open specific issues for this when the time comes.
Thanks again!

pattacini, 2021-07-28:
That's great!
Happy it finally worked.

closed 2021-07-29 by pattacini