robotology/mesh-support #1107: R1 sn001 left arm and right hand malfunction - Luca Nobile

lnobile, 2021-03-24:
## R1 sn001 left arm and right hand malfunction

The left arm (roll joint) and right hand on R1-sn001 have a malfunction.

The roll joint of the left arm doesn't properly calibrate and the yarprobotinterface returns the error:
`2FOC hall sensor fault. Invalid sequence in motor Hall sensors, please check motor hall cable connections. par16 = ID of joint. +.`

The right hand wirst pronosupination joint seems mechanically broken (you can ear the motor spinning but there are no visible joint movements).

fbiggi, 2021-03-24:
right wrist repaired

emilianob80, 2021-03-24:
After the covers were removed, the AEA support of the shoulder was found broken. The helicoil was ripped from the shoulder

[image]

The 5N2 cable that connects the EMS electronic board (EB5) to the crushed and damaged F / T sensor of the left shoulder was also found. The cable has been totally replaced.

[image]
[image]

Regarding the hall sensor, a visual check was carried out and nothing abnormal or damaged was found. We recommend a functional check when it will be possible (given the short notice it was not possible to deepen the debug)

cc: @Fabrizio69 @lnobile @fbiggi @violadelbono

closed 2021-03-24 by emilianob80