robotology/mesh-support #1109: Missing boards (ethernet chain broken)

randaz81, 2021-03-26:
Request for assistance for CER02 SN001 (currently located in Erzelli).
Cannot ping boards 10.0.1.7 and 10.0.1.10 @emilianob80

emilianob80, 2021-03-26:
[image]

emilianob80, 2021-03-26:
@violadelbono probabilmente il problema lo risolvi verificando :  

cavo eth E6 in uscita da scheda 10.0.1.6 su wrist 

cavo eth E6 in ingresso su scheda 10.0.1.7 su mano

da FirmwareUpdater si vede chela catena eth è collegata fino alla 10.0.1.6 compresa

[image]

cc: @randaz81

violadelbono, 2021-03-31:
Cable eth 6 was broken at the exit to board 10.0.1.7
This is due to the sliding of the electrical cables inside the cover since there is a very little space. 
The problem has been solved by restoring the cable and putting zip ties in order to flatten the cables as much as possible.

@randaz81 please open a new issue for the shoulder problem. Thanks!

closed 2021-03-31 by violadelbono