robotology/mesh-support #1110: R1 sn003 neck_pitch_joint malfunction

lnobile, 2021-03-29:
## Description

The R1 sn003 neck_pitch_joint doesn't move: when I try to give a position command the joint doesn't actuate.

## Detailed context

yarprobotinterface doesn't return any error.
Encoder readings seem to be fine.
From yarpmotorgui, when a command is given, the encoder reading box gets coloured in red and the head doesn't move.

pattacini, 2021-03-29:
@lnobile please don't assign people yourself 😃 
We'll take care of doing triaging.

emilianob80, 2021-03-31:
@lnobile all motors of the head are working properly. I have not found any hardware anomalies. The only problem that can be there is the burnt channel of the electronic board. I ask you to please check the operation of the motors again before eventually replacing it

cc: @violadelbono @Fabrizio69

emilianob80, 2021-03-31:
> @emilianob80
> I checked again and from the yarp motor gui I'm not able to move the joint, I'm able to control the other joint of the head so I don't think it's a configuration issue. I also checked the yarprobotinterface configuration files and everything seems ok.

@lnobile at this point I suspect the channel of the electronic board. I can try to program one and replace it in order to remove the doubt. I have not found other hardware problems

emilianob80, 2021-03-31:
[image]
@lnobile I replaced the electronic board and put it in a temporary way in order to do a test run. If the problem is solved la.momto permanently

Cc: @Fabrizio69

lnobile, 2021-04-01:
Thanks, we tested the board and updated the firmware to the latest version. Everything seems to work correctly.

emilianob80, 2021-04-02:
@lnobile Tuesday 06/04/2021 I will definitively fix the electronic board and apply a ceramic heatsink on it

cc: @Fabrizio69 @maggia80

closed 2021-04-02 by emilianob80

emilianob80, 2021-04-02:
https://github.com/icub-tech-iit/tickets/issues/1158#event-4544986603