robotology/mesh-support #1111: Right arm detects external fault pressed, but it is not

S-Dafarra, 2021-04-01:
## Brief description of the request/failure
The right arm prints an error about the external fault pressed, but it is not. Only the the thumb and the index appear to be in idle though.

[image]

## Detailed context

## Accompanying material

Here the log 
[log_icub-head_yarprobotinterface_3349_external_fault_false_positive.txt](https://github.com/robotology/icub-tech-support/files/6242143/log_icub-head_yarprobotinterface_3349_external_fault_false_positive.txt)

maggia80, 2021-04-07:
@violadelbono it should be the fault connector on the board EB28 that is mounted here:

[https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub2.5/iCub%20wiring%20Rev3.3%20COM-EXP%20and%20MC4-PLUS%20and%20BatteryPack/pdf/Motor%20and%20board%20placement_iCub2.5.pdf](https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub2.5/iCub%20wiring%20Rev3.3%20COM-EXP%20and%20MC4-PLUS%20and%20BatteryPack/pdf/Motor%20and%20board%20placement_iCub2.5.pdf)

the connector is P6:

[image]

violadelbono, 2021-05-24:
Solved after the general check of the arms. Closing

kouroshD, 2021-05-28:
Following the recent experiments on the robot, we noticed the same issue with the right arm. Here is the further explanation:

After the assembly has been performed, we have the following problem with the right arm and hand:

the right thumb oppose encoder feedback is high, so it is stays in HW fault

[image]

We face a couple of times right elbow was not configuring.
This is the image for the right prosup, however, later we fault the robot and unfault the robot, and restrarted the robot and it started working. However, for the sake of reporting, I am mentioning it here:

[image]

An issue was closed about this recently, in https://github.com/robotology/icub-tech-support/issues/1111

I can say also the right wrist prosup is not working as expected:
In this case, we were moving the right prosup to the whole possible range of motion using yarp motor gui, while in reality it was not moving more than 20-30 degrees. You can see the robot behavior in the video attached here.

[image]

Moreover, in the calibration process the right thumb oppose reaches `0` degrees and therefore cannot smoothly calibrate the right thumb and touches the hand. Instead, the right thumb reaches the limit of 12 degs.

Here is a video showing when we try to reach all the range of the robot right wrist prosup motion using motorgui while in reality the robot range of motion was much less:

https://user-images.githubusercontent.com/17707730/119855063-e444de80-bf11-11eb-9b2c-999f331c5d19.mp4

cc @DanielePucci  @violadelbono

violadelbono, 2021-06-03:
This is hopefully solved by:

- Having re-crimped all the FAULT connectors of the three mc4plus boards (EB 27-28-29)
- Re-soldered a left elbow encoder cable since it was loose and detached, causing AEA invalid data reading. Since the forearm cover is broken at the level of the screw attachments, I think it moved and hit the sensor cables.

The pronosupination problem is probably related to https://github.com/robotology/icub-tech-support/issues/673

closed 2021-06-03 by violadelbono