robotology/mesh-support #1112: Spikes in left arm encoders

S-Dafarra, 2021-04-01:
## Brief description of the request/failure
While trying the yoga, the robot fell once because of spikes in the encoders of the left arm.

## Detailed context

## Accompanying material

[log_icub-head_yarprobotinterface_3349_external_fault_false_positive.txt](https://github.com/robotology/icub-tech-support/files/6242369/log_icub-head_yarprobotinterface_3349_external_fault_false_positive.txt)

S-Dafarra, 2021-04-01:
Note that the log is the same of #1111

violadelbono, 2021-04-09:
The encoder with spikes is the AEA of joint 6 - left wrist. Probably some cable loose.
I will check board EB24 connector P11.

[image]

kouroshD, 2021-05-28:
As I mentioned to @violadelbono , we have the following problems:
- the left hand wrist encoder does not change values,
- Moreover, we face a couple of times right elbow was not configuring 

[image]
[image]
This is the image for the right prosup, however, later we fault the robot and unfault the robot, and restrarted the robot and it started working. However, for the sake of reporting, I am mentioning it here:

[image]

kouroshD, 2021-05-28:
I can say also the right wrist prosup is not working as expected:
In this case, we were moving the right prosup to the whole possible range of motion using yarp motor gui, while in reality it was not moving more than 20-30 degrees. You can see the robot behavior in the video attached here.

[image]

Here is a video showing when we try to reach all the range of the robot right wrist prosup motion using motorgui while in reality the robot range of motion was much less:

https://user-images.githubusercontent.com/17707730/119855063-e444de80-bf11-11eb-9b2c-999f331c5d19.mp4

cc @DanielePucci  @violadelbono

maggia80, 2021-05-28:
[image]

the AEA sensors for the wrist are LA-AE5 and LA-AE6

we should check the connectors P10 and P11 of the board EB24

[image]

violadelbono, 2021-06-03:
Both the connectors of the 2 encoders were loose and  have been re-crimped. The encoder cable of the left arm was also replaced.

closed 2021-06-03 by violadelbono