robotology/mesh-support #1113: Error in hall effect sensor of left shoulder roll

randaz81, 2021-04-01:
yarprobotinterface detects a anomaly in the sequence of the hall effect sensor of the left shoulder roll motor of R1 - CER02

@violadelbono tried to swap the cable with the other channel of the board (shoulder pitch) and the error is now detected on this new channel. This confirms that problem is not on the board. Thus, the fault appears to be either in the sensor cable or inside the motor.

Additional note:  connecting the test joypad, the following message appers: "hall effect sensor error". My suggestion is to improved the displayed message showing also the internal status  of the hall sensor (e.g. 001, 010 etc), so that it is possible to get additional infos on the fault (e.g. identify if the sensor is properly powered, or if just one of the three channels detect which of the three sensors/cables is broken.

violadelbono, 2021-04-06:
We can try to replace the motor:

| IITCODE  | ALIAS  |  Description | 
|---|---|---|
|  10129 | MECAPION_APM-SA01ACN-8  |  Framed brushless motor 40mm flange, 100W, 3000rpm, incremental 2048 P/R, BEMF 8, Nominal torque 0.318 Nm | 

However, it seems there is no stock in the warehouse at the moment.

[image]

We might need to order it.
cc @Fabrizio69 @maggia80

maggia80, 2021-04-06:
we could check if we have a spare one in some of the setups we build in the past (ask massa elio or ale scalzo) , otherwise, we buy a couple of them.

violadelbono, 2021-04-14:
Motor replaced with a spare one with the help of @fbiggi 🎉 
Now roll joint is working.

closed 2021-04-14 by violadelbono